evilsocket/pwnagotchi #618: Reduce SD Writes by moving .png to a tmpfs mount

Shadyman, 2019-11-16:
**Problem**

Whenever /root/pwnagotchi.png is written, it is written to the SD card, which has a limited write lifespan. Large quantities of writes can be averted by creating the Pwnagotchi display in RAM instead.

**Preferred Solution**

1. Create an fstab entry for a temporary folder (Location TBD)
2. Change pwnagotchi code to create the .png in the temporary folder
3. Create a symlink from the current location (_/root/pwnagotchi.png_) to the temporary folder's pwnagotchi.png

Thank you for your consideration.

benleb, 2019-11-16:
I think we should do this with systemd tmpfiles.d, seems the way to go these days.... https://www.freedesktop.org/software/systemd/man/tmpfiles.d.html

virtadpt, 2019-11-16:
Why not just throw it into /tmp like everyone else?

Shadyman, 2019-11-17:
> Why not just throw it into /tmp like everyone else?

This was going to be my original requested solution, but on the default image, /tmp/ isn't mounted as tmpfs and would be no different than the current location with respect to storage vs RAM, unless I'm misinterpreting the mount list.

I decided against requesting /tmp as tmpfs specifically as I'm not sure how much gets loaded there normally vs the amount of RAM it would have to take up, so I left the location up for debate. Something like /tmp/pwnagotchi could be mounted in /etc/fstab as tmpfs and simply be the home to the PNG file and other temporary pwnagotchi files that aren't required to be "safe" across reboots.

mrseeker, 2019-11-18:
I concur with this: Writes to RAM are less costly than writing to disk, and increases lifespan of the SD card. Things like storing state, temporary files like faces, etc. are better stored in RAM than disk.

xenDE, 2019-11-18:
is there already a possibility to turn off the writing of the LOG file?

evilsocket, 2019-12-07:
agreed, i'll come up with a list of stuff we can keep on tmpfs in ram

PieBru, 2019-12-11:
I would simply use `/run` instead of `/root` 
Here it is on my Pi0 with a fresh pwnagotchi 1.4.1 just installed:
```
pi@pwnagotchi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  3.7G   25G  14% /
devtmpfs        236M     0  236M   0% /dev
tmpfs           240M     0  240M   0% /dev/shm
tmpfs           240M  3.4M  237M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           240M     0  240M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   66M  187M  27% /boot
tmpfs            48M     0   48M   0% /run/user/1000
pi@pwnagotchi:~ $ ls /run
agetty.reload  crond.reboot       log           screen           sudo
alsactl.pid    dbus               motd.dynamic  sendsigs.omit.d  systemd
avahi-daemon   dhclient.eth0.pid  mount         shm              tmpfiles.d
blkid          dnsmasq            network       sshd             udev
console-setup  initctl            resolvconf    sshd.pid         user
crond.pid      lock               rngd.pid      sshwarn          utmp
```

PieBru, 2019-12-11:
On a second Rpi0 I'm using also `log2ram` configured with `ZL2R=true`
```
pi@pg2:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        30G  3.7G   25G  14% /
devtmpfs        236M     0  236M   0% /dev
tmpfs           240M     0  240M   0% /dev/shm
tmpfs           240M  3.4M  237M   2% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           240M     0  240M   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   66M  187M  27% /boot
/dev/zram0       93M  2.2M   84M   3% /var/log
tmpfs            48M     0   48M   0% /run/user/1000
```

mrseeker, 2019-12-11:
If you want to use something like "log2ram", I would recommend splitting the log up into 3 parts:
1. Error messages (File)
2. Debug message (RAM)
3. Session info (File)

That way, if something happens with the pwnagotchi while walking outside, you can easily reproduce the error without having to hotplug it first in your laptop. Saves some frustration when the pwnagotchi crashes ;)

dadav, 2020-04-12:
Derek close

closed 2020-04-12 by derek[bot]