evilsocket/pwnagotchi #621: Allowing Handshakes and Plugins to be stored in the FAT partition

RChadwick7, 2019-11-16:
For those of us who don't use Linux as our main OS, it would be a huge feature to know if plugins can be put on the FAT partition, and if the handshakes could also be stored or copied there as well.

Arttumiro, 2019-11-16:
Biggest problem with that is space on the FAT partition, easy way to get files off and on to the pi is just connecting usb and using sftp to download/upload files. Something like WinSCP if you are a noob like me

RChadwick7, 2019-11-16:
Thanks for the response. Unfortunately, I can't get Putty or WinSCP working. I have a few other headless RPi0W projects that work fine with Putty, and I can access the Pwnagotchi web interface through 10.0.0.2, but not Putty or WinSCP on Pwnagotchi. Maybe I'll try another laptop.
As for the partition, couldn't it be resized? I had to fight to find the smallest MicroSD card, which was a 16GB, which is way overkill for this. A 2GB FAT partition wouldn't be a sacrifice.

zeroSocket, 2019-11-16:
I've just tried FileZilla Client and it's work without any touch on pi...
Host: pwnagotchi  or name in config.yml
User: pi
Password: raspberry or your ssh password if changed
Port: 22
More simple than reboot the laptop with a USB live distro or use a virtual machine

RChadwick7, 2019-11-17:
I just tried FileZilla, but I get - Error: Network error: Software caused connection abort"
 I get similar errors in WinSCP and Putty. Yet a browser to http://10.0.0.2:8080/ shows the Pwnagotchi screen, so the network must be working. I'm glad it's working for you.

mrseeker, 2019-11-18:
For filesharing (FileZilla etc) use SFTP on port 22. Also check if you can connect to that port with Putty. Note though, it takes around 3-5 minutes for the pwnagotchi to fully boot...

pingywon, 2019-11-18:
> I just tried FileZilla, but I get - Error: Network error: Software caused connection abort"
> I get similar errors in WinSCP and Putty. Yet a browser to http://10.0.0.2:8080/ shows the Pwnagotchi screen, so the network must be working. I'm glad it's working for you.

I would be willing to bet your problem lies in the internet connection sharing. Do you have the grid enabled? is you pwnagotchi reporting? Are you setting a static IP of 10.0.0.1? are you able to ping 10.0.0.2? The docuemntation on pwnagotchi.ai for setting up ICS is very good.

RChadwick7, 2019-11-20:
Well, I never got the last version working, but I just upgraded to the latest, and I can SSH fine. :)

RChadwick7, 2019-11-22:
Still, files on the FAT32 partition would make life a lot easier.

zeroSocket, 2019-11-22:
Yes, but on Windows systems because Linux / MacOS supports access to both SD partitions.
You're interested to make backup of pwnagotchi directly from Windows without using CLI?

RChadwick7, 2019-11-24:
Yes, but not just backups. I'd like to easily get handshakes off the Pwnagotchi, as well as adding and editing plugins.