evilsocket/pwnagotchi #622: Allow pwnagotchi to learn/optimize personality based on current speed (GPS)

Shadyman, 2019-11-18:
**Proposed feature**
Allow the speed, as reported by GPS, influence the speed of TTL, Recon, etc and perhaps implement AI to learn the best values for each with respect to speed. 

It could involve something similar to the _"recon_inactive_multiplier"_ in the AI's personality, such as something like a _"speed_divisor"_ which shortens/eliminates sleeping time, channel hops, etc to attempt to capture more packets, deauth more, or whatever the little pwnagotchi decides is best.

**Alternative implementation**
Alternative could involve simple speed-based math and discrete values as described below

**Additional context**
The Wigle android app, for context, has settings for the time between "scans" when you are doing these, which could be used as speed categories if the speed itself isn't calculated into the AI:

- Standing still
- Slower than 8km/h or 5mph
- Faster than 8km/h or 5mph

(This is, of course, understanding that Wigle simply looks for AP, BT, and Cellular signals and not packets, deauth, or implement AI, but could be used as a baseline for feature implementation)

mrseeker, 2019-11-18:
Basically saying, expand the input of the brain with an extra input node called "average_speed", which defaults to 0? That way you can let the brain decide what to do with that information.

Shadyman, 2019-11-18:
Probably based on a rolling average of the last X readings to average out the errors inherent in GPS positioning.

I haven't yet researched much into the A2C brain, so I'm not sure what exactly the implementation would require, but that sounds like a decent start.

xenDE, 2019-11-18:
https://github.com/evilsocket/pwnagotchi/issues/367

mrseeker, 2019-11-18:
Similar, but not same. That one is based on mems sensors and "actual" values at that exact moment. I was more thinking about a history-based version: You use the output of the last loop (average speed) as the input for the next loop. Yes, might break the brain since you will introduce a new input.

mrseeker, 2019-11-18:
When thinking about this, "breaking the AI" might also be an issue if you want to introduce new features to the brain, so #590 might be a dependency for this.

xenDE, 2019-11-18:
yes, i think brain-v2 should be a thing but needs to be tested and have some free/unused values for future/own implementations?

add: for sensors i would use avg too ;)

evilsocket, 2019-11-18:
there's a lot of stuff that can be added as an input, the main issue is that as it is, changing the inputs would break backwards compatibility, meaning we should all delete our /root/brain.nn and retrain one from scratch ... so, before introducing any change whatsoever in the inputs, I want to be able to do this https://github.com/evilsocket/pwnagotchi/issues/590 ... the idea is that, whenever the shape of the neural network changes (thus breaking backwards compatibility), the new root.nn is bootstrapped with the previously accumulated experience in order not to lose it