evilsocket/pwnagotchi #624: pwnagotchi as TOR node

damianchoc, 2019-11-18:
what about making pwnagotchi a TOR node when connected to the internet?

mrseeker, 2019-11-18:
TOR exit node? I rather don't have my pwnagotchi do that...

damianchoc, 2019-11-18:
> TOR exit node? I rather don't have my pwnagotchi do that...

i had only middle nodes on mind

andrewbeard, 2019-11-18:
Considering most of the pwnagotchis connected to the Internet are using a cellular connection I feel like this would probably be more trouble than it's worth. You'd be chewing through a ton of bandwidth (and battery life). Check the relay requirements here:
https://trac.torproject.org/projects/tor/wiki/TorRelayGuide

They include:
- A non-exit relay should be able to handle at least 7000 concurrent connections
- It is required that a Tor relay be allowed to use a minimum of 100 GBytes of outbound traffic (and the same amount of incoming traffic) per month
- A <40 Mbit/s non-exit relay should have at least 512 MB of RAM available.

mrseeker, 2019-11-19:
Just say that you put a Tor client on the raspberry, it's basically dead weight. the pwnagotchi is designed to run offline without any cellular connection. Also, most networks can detect when using tor networks, something you don't want to have with a sniffer. NMAP for example would be a better option than Tor...

damianchoc, 2019-11-21:
> Considering most of the pwnagotchis connected to the Internet are using a cellular connection I feel like this would probably be more trouble than it's worth. You'd be chewing through a ton of bandwidth (and battery life). Check the relay requirements here:
> https://trac.torproject.org/projects/tor/wiki/TorRelayGuide
> 
> They include:
> 
> * A non-exit relay should be able to handle at least 7000 concurrent connections
> * It is required that a Tor relay be allowed to use a minimum of 100 GBytes of outbound traffic (and the same amount of incoming traffic) per month
> * A <40 Mbit/s non-exit relay should have at least 512 MB of RAM available.
> 
> In all seriousness I don't think the TOR network would benefit from having pwnagotchis as nodes of any type.

you are right

closed 2019-11-21 by damianchoc