evilsocket/pwnagotchi #626: [BUG] 1.3.0 Pwnagotchi crashes - TypeError: string indices must be integers

hackabean, 2019-11-18:
My pwnagotchi just got updated to 1.3.0 and it seems like there is an error in:

```
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 123, in <module>
    utils.setup_logging(args, config)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/utils.py", line 186, in setup_logging
    filename = cfg['path']

TypeError: string indices must be integers
```
This can be reproduced when starting the Pi and letting the pwnagotchi service crash or by starting the launcher manually.

hackabean, 2019-11-18:
Non issue

Arttumiro, 2019-11-18:
You should still keep it open to get it fixed ^^

hackabean, 2019-11-18:
I just thought it was a user error. But can do. Thanks @Arttumiro 1.3.0 rocks!!!!

Arttumiro, 2019-11-18:
Hmm, i dont have this issue, maybe it was just user error but i really dont know how tbh

evilsocket, 2019-11-18:
I forgot to mention in the release notes, also the logging changed https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/defaults.yml#L134

evilsocket, 2019-11-18:
Either remove your "main.log" config.yml entry or you make it look like:

```yaml
main:
    log:
        # file to log to
        path: /var/log/pwnagotchi.log
        rotation:
            enabled: true
            # specify a maximum size to rotate ( format is 10/10B, 10K, 10M 10G )
            size: '10M'
```

hackabean, 2019-11-18:
Thanks for answering, new log has been regenerated and we are all good. After that I have restored from backup.

closed 2019-11-18 by hackabean