evilsocket/pwnagotchi #631: gpio_button plugin no longer working[BUG]

miotislucifugis, 2019-11-19:
I just upgraded two units to 1.3.0 and afterward the gpio buttons no longer work on either of them.  Using default configuration using pins  20 and  21 for restart and shutdown.   Ive double checked config.yml and both have gpio_button enabled and those pins/actions listed.   
from my config.yml: 
`gpio_buttons:
      enabled: true
      gpios:
      - '20': sudo touch /root/.pwnagotchi-auto && sudo systemctl restart pwnag$
      - '21': shutdown -h now`
 
Checking the logs, i dont see any indication that the button presses were sensed.

- Pwnagotchi version -1.3.0,  RPI0W, waveshare v2
best

xenDE, 2019-11-20:
works here with fresh burned img 1.3.0. remove the ' around the numbers?

working:

```
        gpio_buttons:
            enabled: true
            #The following is a list of the GPIO number for your button, and the command you want to run when it is pressed
            gpios:
              #20: 'touch /root/.pwnagotchi-auto && systemctl restart pwnagotchi'
              #21: 'shutdown -h now'
                13: 'amixer set Master 3%- ; aplay /etc/pwnagotchi/src-cloned/pwnagotchi-plugin-sound/sound/default/vol.wav'
                26: 'amixer set Master 3%+ ; aplay /etc/pwnagotchi/src-cloned/pwnagotchi-plugin-sound/sound/default/vol.wav'
```

miotislucifugis, 2019-11-20:
yep.. user error here.   For the record, it was the  dashes  before each pin number that was causing . the problem.  Ive always had them there tho, curious that it would be an issue now.   C'est la vie
thanks

closed 2019-11-20 by miotislucifugis