evilsocket/pwnagotchi #633: [BUG] Pwnagotchi trying to push handshakes multiple times

mrseeker, 2019-11-19:
**Describe the bug**
When the pwnagotchi detects an internet connection, it tries to upload handshakes. When the website is not responsive, this results in the pwnagotchi doing an accidental DDoS attack and blowing up the log. The screen also starts flipping and shows false information (2/26 sent, 4/28 sent)

**To Reproduce**
Steps to reproduce the behavior:
1. Get some handshakes
2. Point the config in such a way that it uploads handshakes to a non-existing website (or a non-responsive website)
3. Check the logging

**Expected behavior**
When the upload of any handshake fails, a longer cooldown is needed before attempting again (eg 5 minutes).

**Environment (please complete the following information):**
 - Pwnagotchi version - 1.3.0
 - OS version - Default
 - Type of hardware - Raspi Zero W
 - Any additional hardware used - Wavescreen2

**Additional context**
This seems to only happen when the website is not responding, a "failure to connect" occurs after 30 seconds.

evilsocket, 2019-11-20:
The website is always responding and so is the API (you can check here https://pwnagotchi.ai/map/) ... i think sometimes it's the connectivity that prevents it from connecting ... what is the line that's filling the logs?

mrseeker, 2019-11-22:
"Connection time-out while sending handshake to wpa-sec.stanev.org" repeated multiple times. Also multiple handshakes being uploaded at the same time. Might be caused by "on_internet_available()" being called multiple times in a row...

dadav, 2019-11-22:
The wpa-sec plugin adds the handshakes which got an error to the skip-list. So, you should see the connection-timeout only once per handshake. Isn't that the case?

The plugin "assumes" that the connectivity is given, because, well...on_internet_available is called :D

mrseeker, 2019-11-22:
What I see is a case of "threading" causing the issue: on_internet_available seems to be called multiple times, which causes multiple connection attempts at the same time. At one moment I had 5 simultaneous upload attempts before the website dropped...

maxxer, 2019-11-22:
I'm having the same, mostly on the pi3 but also on the pi0. On the pi3 is more often because I seems to have BT problems on that device, so it often disconnects from internet, and from the WebUI I've seen up to three uploaders running *together* (different total number of nets to be reported).

mrseeker, 2019-12-01:
After a lot of issues I found out that the offending line might be this one:

https://github.com/evilsocket/pwnagotchi/blob/9b594f7fb234d01808cc0ba286a51e8af056c9df/bin/pwnagotchi#L42

As soon as this one gets triggered (every 5 seconds), it boots up another handshake upload...

maxxer, 2020-01-02:
If it's worth the same happens for other plugins as well. I was reading the logs during an automatic upgrade and while pwnagotchi issued reboot another instance of the plugin was fetching the release file