evilsocket/pwnagotchi #639: Disable serial

RChadwick7, 2019-11-20:
I have a RPi0W embedded in an enclosure, with a pushbutton power switch outlined here: https://github.com/NeonHorizon/lipopi. The way it works is that on powerup, GPIO14 (Serial TX) goes high, and stays high, which is used to tell the power supply to keep it powered. I would like to run Pwnagotchi on it, but shuts down after it boots up.

Typically, I add these to config.txt:
enable_uart=1
dtparam=uart0=on

In cmdline.txt, I remove this section:
console=serial0,115200

xBelladonna, 2019-12-19:
Unfortunately this is a limitation of hardware of Raspberry Pi 3B and Zero W because bluetooth uses PL011 UART and the UART available on GPIO pins by default is second one called "mini UART" which for some reason stops transmitting and receiving when SPI1 (aux SPI device) is enabled, and pwnagotchi release image enables SPI1 with chip select 3 as dtoverlay in `/boot/config.txt`.

You will have to move the bluetooth and attach to the mini-UART with lower baud rate (and maybe freeze/drop packets here and there), then use PL011 UART as serial on GPIO. You can do this with `dtoverlay=pi3-miniuart-bt` and use `/dev/serial1` instead of `/dev/serial0`.

Or if you don't need bluetooth you can disable with `dtoverlay=pi3-disable-bt` and just use `/dev/serial1` like above.

RChadwick7, 2019-12-19:
Thanks for the response, but I think you might misunderstand me. I'm not trying to communicate by serial over GPIO14, I am trying to use a property of the Pi where, if it's enabled but not used, GPIO14 goes high when the Pi powers up and stays up until it shuts down. I'm using that property to create a soft power switch so the Pi can power itself down. Unfortunately, it appears Pwnagotchi is messing with GPIO14. When I boot Pwnagotchi on hardware that will shut down when GPIO14 goes down, it boots up for 10-15 seconds and then shuts down. Something is messing with GPIO14. In RasPi-Config, I have enabled Serial, and disabled console over serial. Is Pwnagotchi trying to write to serial?

xBelladonna, 2019-12-19:
Oh goodness I apologise. I am a little confused since GPIO 14 is normally ground, but also can be used for serial in an ALT mode. At this point it is a guess but it could be that maybe some UART is in ALT mode on those pins and that is used by some code, I did not check the whole code base so I don't know more than that I'm afraid.

xBelladonna, 2019-12-23:
Hello again, unfortunately I do not have real answer or solution for you, sorry! 😞
However, here is what I found:
I did not know whether you were referring to "pin 14" or "BCM 14", so I had a look at some datasheets and it seems that BCM 14 (pin 10) is UART TxD in *some* mode, I don't know which because the documentation is unclear what is the default state on power-on/bootup.
It doesn't seem like any code from pwnagotchi touches UART transmit.... so it's not pwnagotchi itself, but maybe property of Raspbian in current configuration of the disk image.

At this point I am speculating, but others on [various forums](https://raspberrypi.stackexchange.com/a/1142) report voltage fluctuations on GPIO lines while device is booting, so perhaps this is interfering with your soft-switch? No way to know for sure unless you measure with oscilloscope 😬

RChadwick7, 2019-12-24:
Thanks again. I was thinking I should put it on an oscilloscope. I'll see if I can find time after Christmas and report my findings.