evilsocket/pwnagotchi #643: webcfg is adding single quotes [BUG]

miotislucifugis, 2019-11-22:
**Describe the bug**
Everytime I save on webcfg, single quotes are added around the gpio pin number definitions under the gpio_button plugin- thus breaking their functionality.   It only seems to be happening with the gpio entries.  I can manually delete the quotes the old fashioned way and it will work again... until I use webcfg to make any changes.  i was configuring GPS settings.  It seems to do it with any change/save, I wasnt doing anything to gpio_button.

before:
`    gpio_buttons:
      enabled: true
      gpios:
        20: sudo touch /root/.pwnagotchi-auto && sudo systemctl restart pwnagot$
        21: shutdown -h now
`
After saving with webcfg:
`    gpio_buttons:
      enabled: true
      gpios:
        '20': sudo touch /root/.pwnagotchi-auto && sudo systemctl restart pwnagot$
        '21': shutdown -h now
`
Ive deleted the whole entry and copied/pasted it from default.yml with the same results, so im fairly certain it is not formatting/spacing  errors in my config.yml.

**To Reproduce**
Steps to reproduce the behavior:
1. Used nano to delete quotes around gpio pin #s in config.yml
2. Reboot and test- GPIO buttons work.
3. Change a setting in webcfg( and setting), save, reboot
4. test again.  GPIO buttons do not work.
5. Use Nano to check config.yml
6. Quotes added around pin #s.   delete them, save, reboot
7. test.  GPIOs work.
8 repeat....

**Expected behavior**
Webcfg should not be adding quotes around the gpio pin numbers the pwnagotchi boots and works fine without them.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.3.0 
 - waveshare v2
 -ups-lite
macos using firefox for the web browser

Also see [https://github.com/evilsocket/pwnagotchi/issues/631](url)  which i opened and closed yesterday because i thought it was user error.   Apparently not, as I can repeated get it to "break" again with a webcfg save .

xenDE, 2019-11-22:
hi, looks like an bug in the gpio plugin.

can you test this:
add a line after https://github.com/evilsocket/pwnagotchi/blob/master/pwnagotchi/plugins/default/gpio_buttons.py#L34

old:
        for gpio, command in gpios.items():
            self.ports[gpio] = command

new:
        for gpio, command in gpios.items():
            gpio = int(gpio)
            self.ports[gpio] = command

and test it with quoted gpio numbers?

miotislucifugis, 2019-11-22:
Adding that line does the trick!  Everything works perfectly even with single quotes.  Thanks

xenDE, 2019-11-23:
thanks for testing!

closed 2019-11-23 by evilsocket