evilsocket/pwnagotchi #646: webcfg deleted my config.yml

miotislucifugis, 2019-11-23:
I had a failed save while using webcfg and it erased all the content of my config.yml.    
Im not sure exactly what happened; whether if it was a pwnagotchi error, or the browser... but after it was auto reset (after clicking save in webcfg)   the pwnagotchi started acting "weird" and i checked the config.yml via nano and it was totally empty.   I actually thought at first that i had mis-typed something and was creating a new file, but no.  Config.yml was just totally empty

 No biggie, as i had a fairly recent backup handy so it was all sorted out easily  but I just thought I should give everyone a heads up that this can happen.

dadav, 2019-11-25:
Thanks for your report! Will make it more failproof

closed 2019-11-26 by evilsocket

benallard, 2019-12-18:
My cfg just disapeared as well. not fun. :(

RubenGamerPT, 2020-04-05:
mine too...

PR #654 (2019-11-26): Fix/webcfg check for error
Fixes #646