evilsocket/pwnagotchi #647: Pwngrid version in Pwnagotchi 1.3.0

zeroSocket, 2019-11-23:
I noticed that pwnagotchi 1.3.0 was released with pwngrid 1.10.1 although version 1.10.3 was already available for a few days. Do we need to update the pwngrid to 1.10.3?

sayak-brm, 2019-11-25:
My pwnagotchi automatically updated pwngrid while running the 1.3.0 image

zeroSocket, 2019-11-25:
Ok, but updated to 1.10.1 version (in 1.3.0 image) while on github 1.10.3 war already released. That's my point

sayak-brm, 2019-11-25:
Maybe `pwngrid` v.1.10.3 wasn't ready in time for packaging when building the v1.3.0 image. Either way, the `auto-update` plugin will automatically update your `pwngrid` to the latest version.

[image]

closed 2019-12-06 by zeroSocket