evilsocket/pwnagotchi #648: Waveshare 1.44inch LCD HAT

Evg33, 2019-11-23:
Can you add support Waveshare 1.44inch LCD HAT ?
https://www.waveshare.com/wiki/1.44inch_LCD_HAT

python code examples here:  https://www.waveshare.com/w/upload/f/fa/1.44inch-LCD-HAT-Code.7z

Evg33, 2019-12-05:
[image]

Evg33, 2019-12-05:
[image]

Evg33, 2019-12-05:
[image]

Evg33, 2019-12-05:
[image]
black text

Evg33, 2019-12-05:
Oldscool :) (green terminal)
[image]

Evg33, 2019-12-06:
[image]

Evg33, 2019-12-06:
https://github.com/evilsocket/pwnagotchi/pull/694

Evg33, 2019-12-06:
[image]

closed 2019-12-07 by Evg33

Faludokner, 2021-12-10:
Hey, I know thus is years late, but do you mind explaining how do you set a custom background for the pwnagotchi on the hat like that? It's so cool.

Evg33, 2021-12-10:
> Hey, I know this is years late, but do you mind explaining how do you set a custom background for the pwnagotchi on the hat like that? It's so cool.

It was 2 years ago.
Texts with picture is seen worse.

XZCendence, 2022-06-05:
> > Hey, I know this is years late, but do you mind explaining how do you set a custom background for the pwnagotchi on the hat like that? It's so cool.
> 
> It was 2 years ago.
> Texts with picture is seen worse.
> 

Can you explain how it was done tho? also how can we change the color of the screen from green to something else

V0r-T3x, 2022-06-30:
@XZCendence if this can interest you, I just coded a colorized darkmode for the pwnagotchi and a 1.33 waveshare lcd screen ST7789 and I try to make it work on a ST7735S. Now Im actually figuring out how to program a theme manager. 

the link [here](https://github.com/V0r-T3x/pwnagotchi_LCD_colorized_darkmode)

XZCendence, 2022-06-30:
> @XZCendence if this can interest you, I just coded a colorized darkmode for the pwnagotchi and a 1.33 waveshare lcd screen ST7789 and I try to make it work on a ST7735S. Now Im actually figuring out [...]

you already told me about this in the discord lol, my username is the same there

V0r-T3x, 2022-06-30:
haha sorry '^-^ , I just bought a 1.44 hat but it's not working, did you have tried a 1.44 ST7735S with a stuck image of pwnagotchi by curiosity?

KalyanSingh05, 2025-01-15:
@Evg33 how you did this? Can you please explain the steps!