evilsocket/pwnagotchi #651: download and save cracked passwords from wpa-sec on interval

xenDE, 2019-11-25:
it would be nice if cracked passwords from wpa-secwill be downloaded automaticly and stored in handshakes dir as .pcap.cracked in plaintext - like the quickdic plugin does.

so, extend the wpa-sec plugin with downoad+proccess the cracked pasword list.

steps:

download of the passwordlist:
```
GET https://wpa-sec.stanev.org/?api&dl=1
with Cookie | key=3e47******************6535c
```

process the download:
https://github.com/xenDE/pwnagotchi-tools/blob/master/wpa-sec.founds.insert.py

dadav, 2019-11-25:
You want to use the file to display it on the map, right?^^

xenDE, 2019-11-25:
yes.
I already show the passwords in .pcap.cracked created from quickdic plugin.

dadav, 2019-11-25:
Sounds nice, i will add it^^

dadav, 2019-11-25:
Added this: https://github.com/dadav/pwnagotchi/tree/feature/wpa-sec-download

but need to test ^^

dadav, 2019-11-25:
I'd like to keep it raw and let other plugins parse it for themself

xenDE, 2019-11-25:
looks good, but it downloads only if it has a new handshake for upload.

I need to think about "other plugins parse it for themself" - not sure what's the best way.

the potfile is only usable for extract password - implement the parser at every plugin that wants to have the password is not a good way IMHO.

xenDE, 2019-11-25:
is working.
only a log message i miss

dadav, 2019-11-27:
> looks good, but it downloads only if it has a new handshake for upload.
>
> I need to think about "other plugins parse it for themself" - not sure what's the best way.
>
> the potfile is only usable for [...]

well, its one for-loop and not really "parsing" because the data is seperated by ":"^^

```
for line in file:
  bssid,stamac,ssid,pw = line.split(":")
```

Yes, it only downloads if a new handshake is found...I don't want to implement another "tracking" file to track the last check.

dadav, 2019-11-28:
Derek close

closed 2019-11-28 by derek[bot]

agent932, 2019-11-28:
Would there be a way to add the cracked networks to the ignore list so the pwnagotchi stops trying to get new handshakes?

itsdarklikehell, 2020-02-04:
> Would there be a way to add the cracked networks to the ignore list so the pwnagotchi stops trying to get new handshakes?

I second this!☝️

agent932, 2020-02-04:
👍

jacopotediosi, 2020-04-03:
Onlinehashcrack (starting from PR #821) saves cracked passwords into .pcap.cracked files (one for each password), allowing them to be usable with the webgpsmap plugin.
It would be nice if wpa-sec did the same (?)

jacopotediosi, 2024-07-20:
Just made the PR #1248 to save wpasec cracked password into individual .pcap.cracked password as discussed above