evilsocket/pwnagotchi #653: Cant access partition on sd card after flashing the image

dangomushi92, 2019-11-25:
after flashing the image in windows and after booting the rpi zero for the first time i can only access the boot partition.
plus i cant get my screen to work (is a waveshare 2.13 D hat)
https://www.waveshare.com/wiki/2.13inch_e-Paper_HAT_(D)

i tried to reformat and reflash multiple times but i cant get it to work.
pls help ty in advance

dadav, 2019-11-25:
How do you try to access the partitions? Via. windows? The second partition is an ext-partition, which you can't open with windows. If it's not windows (you have a linux machine), try the lsblk comand to display the partition. What do you see? We need more informations^^ With the provided informations there is no way we could help.

dangomushi92, 2019-11-25:
ty a lot im so stupid i didnt realize that the partition was in ext...i used my ubuntu machine and had no problems with that. Now im having issues with the paper screen i listed above. i typed 'waveshare213d' in the screen section in the config file but it still wont turn on. :c. i saw it should be supported after version 1.1.0

dadav, 2019-11-25:
Any chance you can check if your screen is working at all? Like, running the vendors demo projects or something. Also, try to find related errors in the logs (journalctl, dmesg).

sayak-brm, 2019-11-25:
> i typed 'waveshare213d' in the screen section in the config file but it still wont turn on. 🌵 

Try with `waveshare_2` in your `config.yml` file.

dadav, 2019-12-07:
Derek close

closed 2019-12-07 by derek[bot]