evilsocket/pwnagotchi #655: Make .pcap files downloadable from web-ui

sayak-brm, 2019-11-25:
**Is your feature request related to a problem? Please describe.**
Copying the handshakes using scp requires me to attach my pwnagotchi to my PC in manual mode and then connect through ssh to copy the handshakes. If the handshake files could be downloaded from the web-ui, I easily could retrieve them on my phone using BT tethering.

**Describe the solution you'd like**
A tab in the web-ui pointing to a dynamically generated page listing the contents of the directory `/root/handshakes` enabling me to download them easily.

**Describe alternatives you've considered**
* *SCP:* Not very convenient.
* *FTP:* Has native apps for every platform, but not very integrated to the rest of the platform (web-ui/pwngrid) if the user has to open a separate app just for the handshakes.

**Additional context**
Also, maybe a tab to port 80 can just be shown when booted in manual mode for easier access to bettercap's web-ui.

sayak-brm, 2019-11-25:
Would deprecate issue #621

dadav, 2019-11-28:
> 
> 
>     * _SCP:_ Not very convenient.
> 

objection!!! or..how to spot the windows user :D

Anyway, I don't rly see me using this kind of functionality, but please feel free to create a plugin for that.

sayak-brm, 2019-11-28:
Actually, I was planning to get the handshakes right off the pi via `bt-tether` on my phone.

Either way, wrote some code, will be testing that ASAP!

closed 2019-12-04 by sayak-brm