evilsocket/pwnagotchi #656: Bettercap does not release GPS port after disconnect[BUG]

miotislucifugis, 2019-11-26:
**Describe the bug**
Ive been messing around with gps both thru USB and bluetooth (a serial gps module +hc05+lipo)  and been having trouble reconnecting if the gps device becomes disconnected.   A usb gps module that was ttyAMC0  becomes now AMC1.  The BT gps module that is communicating on rfcomm0 can not be released or reconnected because it is already in use:
`Can't bind RFCOMM socket: Address already in use`
It takes a reboot to get it working again.

After some digging Ive discovered that bettercap is hanging onto the port after it has been lost  and this(i believe) is preventing it from being released... and available to be reused.
`pi@FeFe:~ $  sudo lsof | grep /dev/rfcomm0 
bettercap  396                            root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  417 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  418 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  421 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  430 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  439 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  458 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  548 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  553 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  560 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396  564 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
bettercap  396 4575 bettercap             root    8u      CHR      216,0       0t0      12323 /dev/rfcomm0
pi@FeFe:~ $ 
`
Only seems to happen (with usp devices at least) after the gps has started sending 'good' data...with rfcomm, it is anytime the bt connection is broken and it never reconnects.

**To Reproduce**
Steps to reproduce the behavior:
1. plug in GPS (usb on ttyAMC0)
2. wait for signal then disconnect, reconnect 
3. ls /dev/tty*   shows the usb device is now connected as ttyAMC1

**Expected behavior**
I would expect the port to be released after it was disconnected

**Environment (please complete the following information):**
 - Pwnagotchi version 1.3.0 w/ waveshare v2
 various usb gps modules, 
  serial gps + HC05 bt

evilsocket, 2019-11-26:
did you try to "gps off" and "gps on" on bettercap?

miotislucifugis, 2019-11-26:
you mean vis ssh?
I just tried disconecting the (bt) gps = cycle power, bt connection is lost
`sudo bettercap`
`gps off` ->  ` » [17:16:02] [sys.log] [err] module gps is not running`  (when using bt serial on rfcomm, it always says that  gps not available, even when it is and is working)
`gps on`->` [17:18:21] [sys.log] [err] open /dev/ttyUSB0: no such file or directory`

Not sure where it is getting USB0.    gps should be  /dev/rfcomm1  as that is what is in my config.yml, right?   Why is it defaulting to /dev/ttyUSB0?   

I havent had time to try this yet with the usb gps modules that connect on /dev/ttyACM0,