evilsocket/pwnagotchi #659: [BUG] Using fahrenheit in memtemp plugin does not display temp properly

gschmall, 2019-11-27:
**Describe the bug

There is not enough room to properly display the working temp in F when setting "scale" to fahrenheit.  

**Solution 

Move the memtemp ui element over a little bit:

 def on_ui_setup(self, ui):
        if ui.is_waveshare_v2():
            h_pos = (170, 80)
            v_pos = (180, 61)
        elif ui.is_inky():
            h_pos = (130, 68)
            v_pos = (165, 54)
        else:
            h_pos = (145, 76)
            v_pos = (180, 61)

pingywon, 2019-12-23:
This bug still exists. Anyway someone can include this in the next round of updates

xenDE, 2020-01-18:
@gschmall @pingywon it's working? then please make a pull request.

moop2000, 2020-10-06:
This does appear to still be a bug. When I switch to fahrenheit, it only shows 2 characters and no symbols. This is on the recommend 2.13" eink display.

meldraknailo, 2020-10-06:
It works for me in the vertical display mode. Try that and let me know if you need me to copy and paste my config files for that. :). It took some tinkering but it displays nicely.![Uploading 95237736...]

meldraknailo, 2020-10-06:
I do recall having to change my position coordinates to show F correctly

meldraknailo, 2020-10-06:
I can confirm it is still a bug in the sense that it does display F but the position coordinates are incorrect and only two digits are displayed. After adjusting the position coordinates it displays correctly. The update should include a default position for F that displays correctly. Here is an image of it displayed correctly after modifying the h and v pos. [image]