evilsocket/pwnagotchi #661: Being able to disable deauth while running

maxxer, 2019-11-27:
**Is your feature request related to a problem? Please describe.**

I'd like to be able to disable and enable `deauth` on runtime.

**Describe the solution you'd like**

It would be great to have an additional button on the webui to enable/disable the feature. It's useful for example if you're entering a meeting and don't want do *disturb*.

Maybe related to #206.

**Describe alternatives you've considered**

Right now the only option coming to my mind is to ssh to the device, change the setting and restart.

dadav, 2019-11-28:
Isn't this solved by `webcfg` plugin? You can change the setting there..

maxxer, 2019-11-28:
Likely, I didn't check for recent plugins before posting 🤦‍♂ 

I'm closing,  will reopen eventually. Thanks!

dadav, 2019-12-16:
Derek close

closed 2019-12-16 by derek[bot]