evilsocket/pwnagotchi #662: Is it possible to run UI via bluetooth tether on android?

H4ck3r404, 2019-11-27:
## Expected Behaviour

## Current Behaviour

## Possible Solution

## Steps to Reproduce (for bugs)

1.
2.
3.
4.

## Context

## Your Environment

- [ ] You're using the official images

- [ ] You're using a raspberry pi 0

- [ ] You're using a supported LCD

H4ck3r404, 2019-11-27:
*default bluetooth config from homepage.

wedemmoez, 2019-11-27:
android bluetooth PAN IPs are generally in the 192.168.44.xx range. try one of those. 10.0.0.2 is the static ip for the USB0 interface and will not work over bluetooth.

H4ck3r404, 2019-11-27:
So what can I do with bluetooth? Trying to connect to (bt-pan ip):8080 or pwnagotchi.local does not seem to work as well as ssh connection.

zeroSocket, 2019-11-27:
Post you complete config.yml. Did you set correctly the ui.web section?

ZeroPoke, 2019-11-27:
> So what can I do with bluetooth? Trying to connect to (bt-pan ip):8080 or pwnagotchi.local does not seem to work as well as ssh connection.

I found out recently that the UI via Bluetooth tethering only works when Im connected to my own WiFi.  If Im out and about I cant see the UI but I can still SSH in.

H4ck3r404, 2019-11-27:
main:
    name: 'pwnagotchi'
    whitelist:
     - 'homewifi'
    plugins:
      bt-tether:
        enabled: true
        devices:
          Redmi:              # you can choose your phones name here
            enabled: true         # enables the device
            search_order: 1       # in which order the devices should be searched. E.g. this is #1.
            mac: 'XX:XX:XX:XX:XX' # you need to put your phones bt-mac here (the same as above, 
                                  ## or goto your phones   settings > status)
            ip: '192.168.44.XX'     # this is the static ip of your pwnagotchi
                                  ## adjust this to your phones pan-network (run "ifconfig bt-pan" on your phone)
                                  ## if you feel lucky, try: 192.168.44.44 (Android) or 172.20.10.6 (iOS)
                                  ## 44 is just an example, you can choose between 2-254 (if netmask is 24)
            netmask: 24           # netmask of the PAN
            interval: 1           # in minues, how often should the device be searched
            scantime: 60          # in seconds, how long should be searched on each interval
            share_internet: true  # this will change the routing and nameserver on your pi
            priority: 99          # if you have multiple devices which can share internet; the highest priority wins 
            max_tries: 0          # how often should be tried to find the device until it is disabled (to save power)
                                  ## 0 means infinity
          macbook:
            enabled: false
            # ...
ui:
    web:
        username: 'login'
        password: 'password'
bettercap:
	handshakes: /handshakes

H4ck3r404, 2019-11-27:
Mac and ip are of course filled with true data. Also when I've added web.ui, bettercap, name and whitelist my device is not visible in bluetooth. (old pair was canceled.)

zeroSocket, 2019-11-27:
Ok, the `macbook` device can be deleted as the `bettercap` section, have you trusted your phone mac in `bluetoothctl`? Anyway... start splitting the problem...
1) Try with a clear config.yml with the only name, whitelist and bt-tether Redmii: check if pwangotchi can pair the phone. Also check if your bt-pan on the phone is really 192.168.44.XX with the `ifconfig` command on your phone terminal (like Termux) and eventually correct the config.yml.
2) Try to connect to 192.168.44.XX:8080 (or the correct ip) with your phone **without** touch config.yml
3) Add the **complete** ui.web section in config.yml (check the default.yml) and see if your bluetooth's phone can still be paired and retry to connect to 192.164.44.XX:8080 (or the correct ip)

H4ck3r404, 2019-11-28:
Does config.yml overwrites old config by reseting  default settings and then making changes?

zeroSocket, 2019-11-28:
Technically config.yml overwrites the settings in default.yml, so yes

H4ck3r404, 2019-11-28:
And how do I trust in phone bluetoothctl without using pwnagotchi terminal?

zeroSocket, 2019-11-28:
Why you can't ssh your pwnagotchi now? There's a lot of confusion...
Post your hardware setup and your config.yml

H4ck3r404, 2019-11-28:
My computer's mainboard is broken and because of that i don't think connecting anything to usb is secure. (even flashing images causes a lot of errors.) 
For now I'm working on windows laptop which can't do the ssh connection properly. My config.yml looks exactly like above. I'm using balena etcher flashing unpacked img file.

zeroSocket, 2019-11-28:
The config.yml must be only one (How is possible create two file with the same name in same directory?), so open the default.yml, copy and past all content in config.yml and edit the part regarding bluetooth, than ssh into RPi (on Windows I use Cmder), pair the phone and trust it, copy config.yml in boot partition and launch.
This procedure is well described in pwnagotchi site, give it a read for more details

H4ck3r404, 2019-11-28:
There were two files - first on the first boot second on the second boot.

Anyway can I connect pwnagotchi by bluetooth on the same device I'm having ssh connection? I mean can I do it on my phone connected by usb with Termux? (phone is rooted.)

zeroSocket, 2019-11-29:
The unique config.yml is placed in `rootfs/etc/pwnagotchi`. If you put another config.yml in the `bootfs` of sdcard I think it will overwrite the old one in `rootfs`.
You can ssh through the bluetooth without using usb, I never tried through usb honestly

H4ck3r404, 2019-12-04:
Copying and modifying text from default.yml to config.yml does not seem to work. I was waiting half an hour to see if pwnagotchi shows up in bluetooth (it didn't.) Idk what's wrong. I'm stuck with pwnagotchi with no screen, bluetooth or ssh output (my computer is not installing drivers for ssh.) Is there anything I can do now?

phr34k0, 2019-12-06:
> Copying and modifying text from default.yml to config.yml does not seem to work. I was waiting half an hour to see if pwnagotchi shows up in bluetooth (it didn't.) Idk what's wrong. I'm stuck with [...]

What I did is connect a usb to microusb to my computer and other end to data microusb input of the raspi 0w, run that linux network share script (since I use linux) and type this in to the terminal

`bluetoothctl
power on
pairable on
discoverable on
system-alias whatever-name-you-want`

then search for that whatever-name-you-want on your phone, connect and pair, then on the computer copy your device mac address and type in `trust the-device-macaddress`

dadav, 2019-12-07:
Please discuss this in the community forum.

dadav, 2019-12-07:
Derek close

closed 2019-12-07 by derek[bot]