evilsocket/pwnagotchi #663: [INFO] waiting for bettercap API to be available ...

pingywon, 2019-11-27:
On first start up bettercap service never fully starts. pwnagotchi gets stuck waiting for it

**To Reproduce**
clean install of 1.2.1 or 1.3.0
very basic config.yml in the boot partition

 main:
  name: 'pwnnamehere'
  whitelist:
    - 'YourHomeNetworkMaybe'
  plugins:
    grid:
      enabled: true
      report: true
      exclude:
        - 'YourHomeNetworkMaybe'

ui:
    display:
      enabled: true
      type: 'waveshare_2'
      color: 'black' 

System boots. Waveshare screen loads.
"Generating keys, do not turn off" comes on the waveshare. Then the system just idles.

Plug into the data port and ssh in and run

journal` -fu pwnagotchi

or

cat /var/log/pwnagotchi.log

and you can see it is full of:
`[2019-07-10 01:58:02,125] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:03,179] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:04,228] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:05,270] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:06,312] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:07,360] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:08,408] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:09,451] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:10,494] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:11,536] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:12,578] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:13,626] [INFO] waiting for bettercap API to be available ...
[2019-07-10 01:58:14,668] [INFO] waiting for bettercap API to be available ...`

one every second

`$ systemctl status bettercap
● bettercap.service - bettercap api.rest service.
   Loaded: loaded (/etc/systemd/system/bettercap.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-07-10 02:00:26 BST; 24s ago
     Docs: https://bettercap.org
  Process: 2301 ExecStart=/usr/bin/bettercap-launcher (code=exited, status=1/FAILURE)
 Main PID: 2301 (code=exited, status=1/FAILURE)`

**Expected behavior**
Expected fully booted working system

**Environment (please complete the following information):**
RPi0w
pwnagotchi image 1.2.1 or 1.3.0
Waveshare 2.13 V2

evilsocket, 2019-11-28:
what about journalctl -fu bettercap ?

pingywon, 2019-11-29:
no interface mon0

`Jul 10 02:19:18 GadgetConnection systemd[1]: Started bettercap api.rest service..
Jul 10 02:19:18 GadgetConnection bettercap-launcher[618]: command failed: No such file or directory (-2)
Jul 10 02:19:18 GadgetConnection bettercap-launcher[618]: WARNING: Terminal colors have been disabled, view will be very limited.
Jul 10 02:19:18 GadgetConnection bettercap-launcher[618]: bettercap v2.26.1 (built for linux arm with go1.13.1) [type 'help' for a list of commands]
Jul 10 02:19:18 GadgetConnection bettercap-launcher[618]: no interface matching 'mon0' found.
Jul 10 02:19:19 GadgetConnection systemd[1]: bettercap.service: Main process exited, code=exited, status=1/FAILURE
Jul 10 02:19:19 GadgetConnection systemd[1]: bettercap.service: Failed with result 'exit-code'.
Jul 10 02:19:49 GadgetConnection systemd[1]: bettercap.service: Service RestartSec=30s expired, scheduling restart.
Jul 10 02:19:49 GadgetConnection systemd[1]: bettercap.service: Scheduled restart job, restart counter is at 5.
Jul 10 02:19:49 GadgetConnection systemd[1]: Stopped bettercap api.rest service..`

pingywon, 2019-11-29:
lol - this pi was misrepresented when sold to me and is not a Rpi0**W**?

evilsocket, 2019-11-29:
sudo ifconfig -a ?

pingywon, 2019-11-29:
Yup - I got beat. No wifi. Thanks for your direct help man.

closed 2019-11-29 by pingywon

evilsocket, 2019-11-29:
👍

TimWachter, 2021-01-31:
Well, that explains why I couldn't get bluetooth working either... That, or somewhere along the way the past 4 years those interfaces broke on this board, but I can remember having issues before. 
[image]