evilsocket/pwnagotchi #666: Webui auth interpreting password with just numbers as 'int' instead of 'str'[BUG]

tabbek, 2019-11-28:
**Describe the bug**
When password defined in config.yml for webui is just numbers, an error occurs when attempting to log in.

**To Reproduce**
Steps to reproduce the behavior:
1. update to 1.3.0
2. Modify config.yml and change pwnagotchi webui password to just numbers
3. restart pwnagotchi service
4. attempt to access webui
5. put in password
6. observe error in pwnlog

**Expected behavior**
interpret any input in config.yml password field as a string, allowing login

**Screenshots**

20:42:56 [ERROR] Exception on / [GET]
(most recent call last):
 File "/usr/lib/python3/dist-packages/flask/app.py", line 2292, in wsgi_app
   response = self.full_dispatch_request()
 File "/usr/lib/python3/dist-packages/flask/app.py", line 1815, in full_dispatch_request
   rv = self.handle_user_exception(e)
 File "/usr/lib/python3/dist-packages/flask/app.py", line 1718, in handle_user_exception
   reraise(exc_type, exc_value, tb)
 File "/usr/lib/python3/dist-packages/flask/_compat.py", line 35, in reraise
   raise value
 File "/usr/lib/python3/dist-packages/flask/app.py", line 1813, in full_dispatch_request
   rv = self.dispatch_request()
 File "/usr/lib/python3/dist-packages/flask/app.py", line 1799, in dispatch_request
   return self.view_functions[rule.endpoint](**req.view_args)
 File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/web/handler.py", line 66, in wrapper
   auth.password):
 File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ui/web/handler.py", line 59, in _check_creds
   secrets.compare_digest(p, self._config['password'])
unsupported operand types(s) or combination of types: 'str' and 'int'

**Environment (please complete the following information):**
 - Pwnagotchi version : 1.3.0
 - Rpi0w
 - Any additional hardware used

**Additional context**
firefox web browser

evilsocket, 2019-11-28:
in yaml strings need to be quoted, try using " or ' to wrap the password.

closed 2019-11-28 by evilsocket

tabbek, 2019-11-28:
My point was more that the python code should interpret whatever was passed as a string, regardless of what yaml thinks it is.