evilsocket/pwnagotchi #667: Waveshare 2.13 Inch e-Paper Display Hat(B)

copysolo, 2019-11-28:
Waveshare 2.13 Inch e-Paper Display Hat(B)

 Hi, I have the following problem.  

I've bought this ePaper display on Amazon

 [https://www.amazon.de/gp/product/B075FR81WL/ref=ppx\_yo\_dt\_b\_asin\_title\_o00\_s00?ie=UTF8&psc=1](https://www.amazon.de/gp/product/B075FR81WL/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) 

, it remains more leery after the installation of pwnagotchi.  I have now tried and found that the configatin waveshare\_2 does not work, I have another display and it only works with the testscript  

https://github.com/waveshare/e-Paper/blob/master/RaspberryPi%26JetsonNano/python/examples/epd\_2in13bc\_test.py  

What do I have to change in the configuration that it works under pwnagotchi.  

Thank you

copysolo, 2019-11-28:
[image] m’y screen

gagna, 2019-11-28:
According to Waveshare documentation this is the three color version

copysolo, 2019-11-28:
> According to Waveshare documentation this is the three color version

yes indeed, on my config.yml whether in waveshare_v1 or waveshare_v2 I have no reports

cdiemel, 2019-11-28:
@copysolo what do you mean by  not work?  like no display at all or just not in color?

copysolo, 2019-11-28:
sorry, yes I do not have a display on my screen. 
[image]

copysolo, 2019-11-28:
I fix my problem with the config by default.  of course for my screen I had to change the color red problem resolved
[image]
[image]

closed 2019-11-28 by copysolo

weizhao11, 2019-12-02:
I have the same problem as you, but I am not sure what is default configuration which your display work well. 
look forward to your suggestion. Thanks a lot.

copysolo, 2019-12-02:
'waveshare_1'
Color: 'red' 
the screen works with this command but the refresh time is very fast to auto mode even with the command fps 0.0 

if I find a solution I will not hesitate to share it.

weizhao11, 2019-12-02:
@copysolo thanks, it work now.  As you mentioned, refresh time is not right.

copysolo, 2019-12-02:
> @copysolo thanks, it work now. As you mentioned, refresh time is not right.

yes the refresh time of this screen is high.  I advise you to use for the moment the interface ui web time to find a solution, to avoid breaking your hardware

weizhao11, 2019-12-03:
https://github.com/evilsocket/pwnagotchi/blob/19973574e794e172948c7b52577a9849d4070ac1/pwnagotchi/ui/view.py#L120
 if       self._config['ui']['fps'] is 0.0  then the function will meet error.   

Another found: In AI mode, there are so many change of main information. So  eink refresh quickly.
Maybe refresh time of this screen might relate with AI mechanism.

copysolo, 2019-12-03:
I will try to take the risk of increasing the refresh time from 0 to 30, then 60 ect ... see if it has an impact

copysolo, 2019-12-07:
> @ Copysolo merci, ça marche maintenant. Comme vous l'avez mentionné, le temps d'actualisation n'est pas correct.

Hello, I tested this
'waveshare213d'
color: 'black'
the refreshing time and less elevat than 'waveshare_1' color: 'red'

copysolo, 2019-12-07:
> @copysolo thanks, it work now. As you mentioned, refresh time is not right.

Hello, I tested this
'waveshare213d'
color: 'black'
the refreshing time and less elevat than 'waveshare_1' color: 'red'

weizhao11, 2019-12-09:
@copysolo  Thanks！I will try it。

Omeliel, 2023-11-17:
> Waveshare 2.13 Inch e-Paper Display Hat(B)
>
> Hi, I have the following problem.
>
> I've bought this ePaper display on Amazon [...]
I had same problem with vers 1.5.5. I have a 2.13-inch e-Paper hat (B) V4. It worked for me. 
Use this 
https://www.leviathansecurity.com/media/building-a-pwnagotchi-with-the-waveshare-213in-display-version-3
flash this https://github.com/DrSchottky/pwnagotchi/releases, put waveshare_3 in config.toml
I think Levi refers to same link on github. Anyway there is another version of PW 1.7 something if you look on GitHub

TheBaum08, 2023-12-21:
which color did you use ? 

I did every step and it didnt do anything.I have this display : https://www.amazon.de/gp/product/B075FR81WL/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1 

What can i do ? 
Should i plug it directly in the data port ?