evilsocket/pwnagotchi #673: HDMI output

agent932, 2019-11-28:
I would like to put in a feature request to have the Pwnagotchi output via HDMI for all the small lcd displays as well as the in vehicle video displays (in dash)

floSoX, 2020-01-31:
Hi!
I make a PoC (with fbi, bash and perl) that you can find here: https://github.com/solution-libre/pwnagotchi-hdmi-viewer
It works on a Raspberry Pi 1 B
Regards,

tsaquet, 2020-02-04:
Works like a charm, super smart :+1:

Xitro01, 2020-09-28:
The PoC doesn't seem to work anymore, probably due to a lot of changes that has been made in the latest version of pwnagotchi. Did someone got this to work with the latest version? Thanks in advance!

floSoX, 2020-09-30:
Hi @Xitro01
I just release a new version with pwnagotchi 1.5 compatibility.