evilsocket/pwnagotchi #675: Recent Activity/Log Overview in Web UI

Shadyman, 2019-11-29:
**Problem**
I am often curious about Pwnagotchi's adventures and sometimes only have access to the Web UI and not SSH, and logs are often hard to read while getting a good picture of what has been done without a lot of _grep_ and _sed_.

**Solution**
A page on the Pwnagotchi Web UI for "Recent" activity, or logs, that displays a list in table format of association frames, captured handshakes, disassociation frames sent, etc from journalctl. 

This could either be saved in a parsed format on the SD card, or simply be generaed on page load or refresh, based on the output of journalctl or related log output.

**Alternatives**
Alterantively, a simple log viewer could be implemented, but would lack the ease-of-use / finesse of a parsed output.

**Context**
This could also integrate requests such as #655 as pictured below. If GPSd is used, column could be added to prepopulate a google maps link with latitude and longitude.

Any saved version of this log, or its MAC/BSSIDs should be obfuscated in accordance with #381

Quick and dirty example:
[image]

sayak-brm, 2019-11-30:
@Shadyman Just created a PR (#679) for my issue #655. You could consider using that for now to download `.pcap` files while a solution for this issue isn't available

RChadwick7, 2019-12-19:
I would love to see a simple log viewer. At the end of the day, I have no idea what it's found. Downloading handshakes is also great, but I don't think it's a replacement for logs.