evilsocket/pwnagotchi #686: Discussion: Defaults

easyMaxi, 2019-12-02:
If this is discussed in another place, show me the way.

The project affects different parties in the application. Of course, there is a legal discussion, but there is also the perspective of users and administrators of affected networks.

I would argue that the default setting for personality-deauth and personality-associate should be false.
I think the assumption that the user is aware of its consequences the first time the tool is started can not be kept.
In order to inform users, this should be explained in the set-up process and explicit action needed on the part of the user. A quick warning on the frequently asked questions about data collection seems a bit small.

To be sure, I am not arguing to not provide the options. This is more of a legal debate and I have no idea about that. I would like to express that the device may do things that are not intended by the user or that can not be estimated and that the project can do more to counteract this.

Evg33, 2019-12-02:
Yes ! deauth must be disabled by default !
It causes problems for others. (Disconnects)

dadav, 2019-12-03:
Deauthing the clients is one of the two main stragegies of this project. If you feel the need to disable it go ahead, but this request won't ever happen (as far as i can tell, lol).

dadav, 2019-12-03:
Woa, this would suck :D But it wouldnt change my opinion.

evilsocket, 2019-12-03:
github is not for discussion about these things, there's a forum if you feel verbose https://community.pwnagotchi.ai/c/feedback-and-ideas

@Evg33 there's a maximum amount of times that a unit can interact with a single BSSID, this cap was introduced exactly to not make the unit annoying.

closed 2019-12-03 by evilsocket