evilsocket/pwnagotchi #691: 200x200, 1.54inch E-Ink display module (Non-B)

AureliaNYT, 2019-12-04:
Is this screen I have (https://www.waveshare.com/1.54inch-e-Paper-Module.htm) supported, and if not, are there any plans to support it?

Evg33, 2019-12-06:
https://github.com/evilsocket/pwnagotchi/tree/master/pwnagotchi/ui/hw/libs/waveshare/v154inch

```
    display:
        # Possible options ....., waveshare154inch,...
        type: 'waveshare154inch'

```

AureliaNYT, 2019-12-06:
Isn't this for the "B" variant?

Evg33, 2019-12-06:
i don't have waveshare154inch

AureliaNYT, 2019-12-06:
I do not think that it's meant for the non-b variant, as it does not draw anything on the e-Paper screen

AureliaNYT, 2019-12-06:
As I see you are a contributor, can you explain to me how to add support for a screen properly? It's in the docs, but there's basically no info on how to do it from start to finish. I tried putting the libs and a new file to support the screen, but when I built the image and put the new screen in config, it said "screen not supported".

Evg33, 2019-12-06:
I do not have this screen for testing.
I expect delivering waveshare_2.

AureliaNYT, 2019-12-06:
I get it, I am just asking for a bit of explanation on how to add a new screen in general.

Evg33, 2019-12-06:
An example can be seen in my PR #694

AureliaNYT, 2019-12-06:
And then I should just build the image using 'make image'?

AureliaNYT, 2019-12-06:
Is there any more steps to this, as when I build the image (followin what you did) I get "screen not supported"

dadav, 2019-12-07:
What you can do is the following: Install the 1.3.0 Image. Go to /usr/local/src/pwnagotchi. Make your changes. Type "pip3 install .". Run pwnagotchi and test it.

AureliaNYT, 2019-12-07:
I tried in multiple ways, did ```pip3 install .``` , but nothing - still getting this:
```
blackpwn pwnagotchi-launcher[1038]: unsupported display type waveshare154inch_v2

AureliaNYT, 2019-12-07:
NVM, got it working. Thanks guys! =)

closed 2019-12-07 by AureliaNYT

nothingbutlucas, 2024-02-03:
Hi @ytcalifax, would you be so kind as to share how you solved it? Because I am trying to add support to this screen and I am encountering the same problem as you.... Thank you!

ytcalifax, 2024-02-03:
Hi @nothingbutlucas! From what I remember, I used the stock code from WaveShare and modified it a little, however I am not sure if this would still work, as quite a long time has passed. Moreover, I no longer have a pwnagotchi so I can't give you my code, since the only location I had it on was my Pi Zero. I'm sorry I can't help you more than this, but hopefully what I said is enough of a pointer for you to implement it yourself.

nothingbutlucas, 2024-02-03:
Yes, I just found [this repository](https://github.com/jayofelony/pwnagotchi-torch) that seems to support the 1.54 inch v2 waveshare
Thanks for replying!