evilsocket/pwnagotchi #692: error loading Wpa-sec.py[BUG]

miotislucifugis, 2019-12-05:
**Describe the bug**
Ive realized that my  pwnagitchi has not been uploading any new handshakes to wpa-sec.stanev.org for about a week or so.  I just noticed this in the log on start up:
`pwnagotchi-launcher[3548]: [2019-12-05 05:03:07,860] [WARNING] error while loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/wpa-sec.py: Expecting ',' delimiter: line 1 column 16434 (char 16433)`
Any ideas what is going on here?   It worked before, after going to 1.3.0.    As far as I know I made no changes to any to the wpa-sec.py files.   My config did get wiped in a webcfg error, but I restored it 

**To Reproduce**
Steps to reproduce the behavior:
1. restart pwnagotchi
2. observe logs

**Environment (please complete the following information):**
 - Pwnagotchi version 1.3.0
 waveshare v2, upslite, 
bluetooth gps connected on dev/rfcomm1

miotislucifugis, 2019-12-05:
update: i removed the file and made a new one and now it works. (shrug)  
curious tho as to what went wrong...

dadav, 2019-12-05:
This happens when the tracking file is corrupted. This should be fixed in #664

dadav, 2019-12-06:
Derek close

closed 2019-12-06 by derek[bot]