evilsocket/pwnagotchi #695: [BUG] pwnagotchi does not boot into MANU mode

ArnCo, 2019-12-06:
**Describe the bug**
Pwnagotchi does not boot in MANU mode, or only once every ~5 attempts.

**To Reproduce**
Steps to reproduce the behavior:
1. Plug the cable in the data port
2. Wait for the pwnagotchi to boot
3. pwnagotchi boots into AUTO mode

**Expected behavior**
pwnagotchi boots into MANU mode

**Environment (please complete the following information):**
 - Pwnagotchi version : v1.3.0
 - OS version : standard raspbian lite image
 - Type of hardware Raspberry pi 0 W v1.1
 - Any additional hardware used 2.7 inch e-paper hat

zeroSocket, 2019-12-06:
Try with another cable, maybe it's a false contact of usb connector

ArnCo, 2019-12-06:
@zeroSocket Unfortunately, I've tried with 3 different cables, same result. It only worked once...

zeroSocket, 2019-12-06:
Try checking `pwnagotchi.log` in `/var/log` if there's something about or with webui button via bluetooth (if you have configured yet) to reboot in `MANU`. Seems to be more an hardware problem than software (maybe usb port was damaged staying in pocket, like mine before UPS-lite)

ArnCo, 2019-12-06:
@zeroSocket It looks like it has something to do with USB3 ports which do not look like working well with the g_ether gadget... I'm closing the issue.

closed 2019-12-06 by ArnCo