evilsocket/pwnagotchi #708: [BUG] WebGpsMap does not parse Timestamps with UTC-timezone encoded as 'Z'

Hadened, 2019-12-09:
**Describe the bug**
The WebGpsMap-Plugin parses the GPS-Timestamps present in the (gps|geo|paw-gps).json-files and brings them to unix-timestamp format.
To do so, it uses the delimiters '.' and '+' to split the microseconds from the rest:
part1, part2, part3 = re.split('\.|\+', date_iso_formated) (in webgpsmap.py #308)

If the system is set to UTC-Timezone (default behavior of my pwnagotchi made from a downloaded SDcard-image), the split (and the following lines correcting the timezone format fail, cause the timezone is formatted as 'Z' instead of '+00:00' (which is perfectly fine according to ISO 8601.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a x.gps.json file with the string {"Updated": "2019-11-28T04:44:46.79231Z", "Latitude": 48.2026005, "Longitude": 16.348880833333332, "FixQuality": "1", "NumSatellites": 4, "HDOP": 3.63, "Altitude": 246, "Separation": 42.1}
2. Start the parsing by accessing /plugins/webgpsmap
3. Check the pwnagotchi.log (produces an split error)

**Expected behavior**
The WebGPSMap also parses timestamps with timezone=UTC and format 'Z'

**Environment (please complete the following information):**
 - Pwnagotchi version 1.4.1
 - RPi-Zero W + Waveshare 2.13" V2
 - GPS-Dongle via USB

Evg33, 2019-12-09:
raspi-config > 4 Localisation Options > I2 Change Timezone

xenDE, 2019-12-09:
thanx for provide a new time format.

i will add it to the testdata at

https://github.com/xenDE/pwnagotchi-plugin-webgpsmap/tree/master/handshakes.gps-map-test

and add a fix in some days, if no one other make a pr.

closed 2019-12-16 by evilsocket