evilsocket/pwnagotchi #709: [BUG] Grid plugin fails to load at startup

James76931, 2019-12-09:
**Describe the bug**
When Pwnagotchi attempts to load the Grid plugin on startup, it displays this error resulting in the plugin failing to load and the Pwnagotchi's PwnGrid page not updating.
```
[2019-12-09 16:13:52,059] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py
[2019-12-09 16:13:52,067] [WARNING] error while loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py: Expecting value: line 1 column 1 (char 0)
[2019-12-09 16:13:52,070] [DEBUG] Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 81, in load_from_path
    load_from_file(filename)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 69, in load_from_file
    spec.loader.exec_module(instance)
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py", line 44, in <module>
    class Grid(plugins.Plugin):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 18, in __init_subclass__
    plugin_instance = cls()
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py", line 53, in __init__
    self.report = StatusFile('/root/.api-report.json', data_format='json')
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/utils.py", line 360, in __init__
    self.data = json.load(fp)
  File "/usr/lib/python3.7/json/__init__.py", line 296, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
```

**To Reproduce**
Steps to reproduce the behavior:
1. Stop the Pwnagotchi service
2. Start Pwnagotchi with sudo pwnagotchi --debug
3. Observe logs
4. Check the PwnGrid page, there should be no new changes

**Expected behavior**
The error should not appear and data should be uploaded to Pwngrid.

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.4.1
 - OS version: Raspbian which comes with the Pwnagotchi iso
 - Pi Zero W
 - Any additional hardware used: Waveshare V2 Display

dadav, 2019-12-09:
The file `/root/.api-report.json` is probably malformed. Correct it or delete it^^ It should fix your problem.

mateuspestana, 2019-12-09:
Following what @dadav said, I restored my .api-report.json backup (which was working  yesterday). Now, the grid plugin is initiating, but the following error appears when I look at the logs:

Dec 09 22:18:23 caramuru pwnagotchi-launcher[361]: [2019-12-09 22:18:23,289] [ERROR] [grid] error while checking pcaps: 'report'

What should I do?

trevorbryant, 2019-12-09:
I experienced the same error as OP. Backing up `/root/.api-report.json`, restarting pwnagotchi service and pwngrid was able to sucessfully recreated `/root/.api-report.json` and begin parsing to report new networks again.

The parsing has started over from the very beginning, but hey it's moving along now.

James76931, 2019-12-10:
Thanks for your responses, it appears that /root/.api-report.json was causing the issue. I deleted it but didn't bother backing up the file as it didn't have any content and the problem was fixed when I re-ran pwnagotchi. :)

closed 2019-12-10 by James76931