evilsocket/pwnagotchi #712: pwnagotchi.service: Main process exited, code=exited, status=127/n/a

yuppchukno, 2019-12-09:
**Describe the bug**
When i boot up the Pi and run journalctl -fu pwnagotchi I receive and endless loop of the log below. 

Dec 09 09:28:18 PCAPxXxSlut pwnagotchi-launcher[546]: /usr/bin/pwnagotchi-launcher: line 10: /usr/local/bin/pwnagotchi: No such file or directory
Dec 09 09:28:18 PCAPxXxSlut systemd[1]: pwnagotchi.service: Main process exited, code=exited, status=127/n/a
Dec 09 09:28:18 PCAPxXxSlut systemd[1]: pwnagotchi.service: Failed with result 'exit-code'.
Dec 09 09:28:48 PCAPxXxSlut systemd[1]: pwnagotchi.service: Service RestartSec=30s expired, scheduling restart.
Dec 09 09:28:48 PCAPxXxSlut systemd[1]: pwnagotchi.service: Scheduled restart job, restart counter is at 8.
Dec 09 09:28:48 PCAPxXxSlut systemd[1]: Stopped pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..
Dec 09 09:28:48 PCAPxXxSlut systemd[1]: Started pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..

In /var/log/pwnagotchi.log This is the last entry before it goes blank. 

[2019-12-09 09:23:52,464] [WARNING] update for evilsocket/pwnagotchi available (local version is '1.3.0'): https://github.com/evilsocket/pwnagotchi/archive/v1.4.1.zip    [2019-12-09 09:23:53,362] [INFO] [update] downloading https://github.com/evilsocket/pwnagotchi/archive/v1.4.1.zip to /tmp/updates/pwnagotchi/pwnagotchi_1.4.1.zip ...     [2019-12-09 09:24:10,962] [INFO] [update] extracting /tmp/updates/pwnagotchi/pwnagotchi_1.4.1.zip to /tmp/updates/pwnagotchi ...                                          [2019-12-09 09:24:17,834] [INFO] sending association frame to Hotspot8002 (22:ad:56:41:80:02 ) on channel 1 [0 clients], -92 dBm...                                       [2019-12-09 09:24:17,852] [INFO] [update] installing pwnagotchi ...                                                                                                       [2019-12-09 09:24:22,948] [INFO] sending association frame to ATTceshQXi (78:96:84:78:7c:b0 ARRIS Group, Inc.) on channel 1 [0 clients], -90 dBm...                       [2019-12-09 09:24:23,072] [INFO] [update] done                                                                                                                            [2019-12-09 09:24:24,758] [WARNING] rebooting ...                                                                                                                         [2019-12-09 09:24:28,909] [INFO] sending association frame to ARRIS-C562 (38:70:0c:f4:c5:60 ARRIS Group, Inc.) on channel 1 [0 clients], -88 dBm...                       [2019-12-09 09:24:33,380] [INFO] sending association frame to NETGEAR96 (28:80:88:4d:6b:23 ) on channel 1 [0 clients], -87 dBm...                                         [2019-12-09 09:24:37,178] [INFO] sending association frame to ATTwjW68Xi (74:8a:0d:82:19:e0 ARRIS Group, Inc.) on channel 1 [0 clients], -84 dBm...

When i check  /usr/local/bin/ There is indeed no pwnagotchi folder. 
pi@PCAPxXxSlut:/usr/local/bin $ ls
crypto    ds1302_get_utc  f2py   f2py3.7   markdown_py  pasteurize  tensorboard
decrypto  ds1302_set_utc  f2py3  futurize  naked        scapy       UTscapy

**To Reproduce**
Steps to reproduce the behavior:
1. ...Power up the device
2. ...?
3. ...Profitless 

**Expected behavior**
I believe somehow the update corrupted the files. It was working fine until it updated. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version V1.4.1
 - OS version
 - Type of hardware Pi 0w
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

dadav, 2019-12-10:
The update obviously failed. Try to update manually via

```shell
cd /usr/local/src/pwnagotchi
git pull
pip3 uninstall -y pwnagotchi
pip3 install .
```

yuppchukno, 2019-12-10:
Thank You, that worked beautifully!

dadav, 2019-12-16:
Derek close

closed 2019-12-16 by derek[bot]