evilsocket/pwnagotchi #716: rpi3 python 3.7.5 tensorflow 1.13.1 not found

X3eRo0, 2019-12-11:
**DEVICE**:    Raspberry Pi 3
**OS**:    Kali Linux 2019.4
**Python**: 3.7.5 (python3) and 2.7.17 (python)

i used pwnagotchi 1.4.1 release source zip
pwngrid and bettercap's latest release

`Collecting tensorflow==1.13.1 (from pwnagotchi==1.4.1)
  Could not find a version that satisfies the requirement tensorflow==1.13.1 (from pwnagotchi==1.4.1) (from versions: )
No matching distribution found for tensorflow==1.13.1 (from pwnagotchi==1.4.1)`

also i couldnt find python37 wheel for tensorflow and armhf
please help

WheresAlice, 2019-12-26:
There's builds of Tensorflow for Arm here that got me a little further - https://github.com/lhelontra/tensorflow-on-arm/releases

There's a bunch of dependencies that still need building though, and that will be slow.

I never got this fully working though, due to other issues around opencv dependencies.  But I was running Raspbian and had hacked it about so much that that's probably my fault.

4p3rtur3, 2020-01-09:
Try 

`sudo python3 -m pip install --upgrade https://storage.googleapis.com/tensorflow/mac/cpu/tensorflow-1.13.1-py3-none-any.whl`

Found that solution here https://stackoverflow.com/questions/38896424/tensorflow-not-found-using-pip

I could deploy that **wheel** successfully on my _Raspberry Pi 3_ running _Kali arm64_ and on my old _Eee PC_ running _Kali i386_.

closed 2022-06-28 by X3eRo0