evilsocket/pwnagotchi #719: [BUG] Pwnagotchi does not initiate association/deauthing as expected in AUTO mode.

Ayakura, 2019-12-12:
**Describe the bug**
pwnagotchi does not initiate association/deauthing as expected in AUTO mode, this issue started happening from 1.4.1 (I have not used 1.4.0). Version <=1.3.1 was fine

**To Reproduce**
Steps to reproduce the behavior:
1. Either boot into AUTO from MANU, or go straight into AUTO from plugging into power
2. Wait for AUTO to start
3. Be disappointed 

**Expected behavior**
After looking around and waiting, supposedly AUTO will start associating/deauthing as usual.

However, there's an endless loop of waiting->looking around->waiting. After 20m of waiting, AI mode would start and the box would function as expected - but only in AI mode.

**Screenshots**
N/A

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.4.1
 - OS version: Windows
 - Type of hardware: RPI0W
 - Any additional hardware used: 2.13in paper HAT

**Additional context**
Please let me know if you need anything from the log

agent932, 2019-12-12:
mine is doing this as well

xenDE, 2019-12-12:
i have the same problem on one device, need to check at weekend on my second pwnie.
both updated vom 1.3 with:
```
cd /usr/local/src/pwnagotchi/
sudo git pull
sudo pip3 install .
sudo reboot
```

agent932, 2019-12-12:
I am re-flashing 1.4.1 and will report back.

dadav, 2019-12-13:
Should be fixed in: https://github.com/evilsocket/pwnagotchi/commit/dbb64e0fabb3df3d6886b075c1079f607a0142e2

This is not included in 1.4.1 yet.

@evilsocket maybe tag 1.4.2?

PR #722 (2019-12-16): no comment
Fix #719
closed 2019-12-16 by evilsocket