evilsocket/pwnagotchi #721: Support for waveshare 2.13 inch b and c variants

moheshmohan, 2019-12-13:
**Is your feature request related to a problem? Please describe.**
Purchased this https://www.waveshare.com/2.13inch-e-paper-hat-b.htm - SKU 13448 and it didnt seems to be working well with pwnagotchi. As per the pwnagotchi site documentation this tricolor display is not recommended

**Describe the solution you'd like**
Based on the code shared by waveshare over variants 2.13 b&c and d. These code could be merged to run 2.13bc display in 2 color mode with fast refresh rates like 2.13d display.

This support can be then added into pwnagotchi as a new display type waveshare213bc

**Additional context**
I have created a pull request based on the solution that i tried to integrate. Details of this is available at https://github.com/evilsocket/pwnagotchi/pull/720

Please find the photo of two test pwnagotchis running the new code (one is not powered on in the pic) as per the above pull request. The display does not seem to be getting frozen over reboots

[image]

moheshmohan, 2019-12-20:
@evilsocket @dadav Can any of you guys see if its a feasible solution? Please let me know

vipx1, 2020-01-08:
[image]
I have a waveshare 2.13" HAT(B) and I can't get to work at all. It has version no. 2.1, I'm not sure if this is relevant. It also has the green sticker on the front the same as yours.

Can you post up your config.yml ui section
Please so I might get my display working with my Pi Zero W. 

Note: I have a working pwnagotchi on a Pi 4 2GB with a waveshare 2.9inch already so I reasonable idea what I'm doing. 

Thank you, 
Stephen.

slayergithub, 2020-03-04:
Hey, just use "waveshare213d" and it'll work. Found it on Reddit for my Tri-Color display. I got the same package you did, but to my delight found that all I needed to do was set THAT as the display name in config.yml.

moheshmohan, 2020-03-10:
> Hey, just use "waveshare213d" and it'll work. Found it on Reddit for my Tri-Color display. I got the same package you did, but to my delight found that all I needed to do was set THAT as the display name in config.yml.

But after sometime the display starts to freeze so i merged new code to fix this issue.

Now you can use "waveshare213bc" and it will work without getting frozen

slayergithub, 2020-03-10:
Oh okay, I haven’t experienced that but I’ll take your word for it.  Have you figured out how to incorporate a theme so we can use all three colors?

BlackFrog1, 2020-06-07:
@moheshmohan I am just got my pwnagotchi and I had ordered the wrong waveshare213b by accident.  Thanks for submitting this code.  My device is running with v1.5.3 and everything look good.  This ticket can be closed.

ShagoY, 2020-10-26:
Remember to remove the line "ui.display.color" from your config.toml for "waveshare213bc" to work.

ntnz, 2020-12-18:
Hey I've got a waveshare213b and I'm using `ui.display.type = "waveshare213bc"` and `ui.display.color = "black"` with the latest `1.5.3` release. I'm having an issue where the display flashes between, solid black, a black border, and blank (I've also tried the `1.5.1` and it has the same issue). It seems to be looping between these three settings every couple of seconds and never displaying the correct pwnagotchi display. Is this a known issue with the waveshare213b?

KennyNova, 2022-04-02:
@ntnz hey i'm having the same exact issue did you ever find a solution?

5h4rk-lab, 2022-04-19:
ui.display.enabled = true
ui.display.type = "waveshare213bc"
ui.display.color = "black"

this worked for be but yeah It gets refreshed a lot which leads a heavy time consumption.

mrmand77, 2023-04-02:
hello i have tried building a pwnagotchi i have bhought the raspberry pi zero and display three colour red black and white i have tried putting the display name as waveshare213b its still not working everytime the raspberry pi zero flashes 6 times and stops and flashes in a pattern way please help me thank you for your time and consideration

5h4rk-lab, 2023-06-02:
I had the same issue but it took me around 30mins to boot properly. so just leave it on side until it boots up completely.

5h4rk-lab, 2023-09-04:
No not yet still the same problem