evilsocket/pwnagotchi #728: [BUG] Waveshare 2.9 not recognized

ack202, 2019-12-14:
**Describe the bug**
Whenever the display is enabled in the config the service is stuck during startup. See below:

```
[2019-07-10 02:35:07,006] [DEBUG] loading plugins from /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default - enabled: ['grid', 'auto-update', 'led', 'session-stats']
[2019-07-10 02:35:07,007] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py
[2019-07-10 02:35:07,008] [DEBUG] loaded plugin grid as <grid.Grid object at 0xb50cd610>
[2019-07-10 02:35:07,008] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2019-07-10 02:35:07,009] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0xb50cd770>
[2019-07-10 02:35:07,009] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2019-07-10 02:35:07,280] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0xb50cdb90>
[2019-07-10 02:35:07,281] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/led.py
[2019-07-10 02:35:07,282] [DEBUG] loaded plugin led as <led.Led object at 0xb4ef6350>
[2019-07-10 02:35:07,284] [WARNING] ui.fps is 0, the display will only update for major changes
[2019-07-10 02:35:07,284] [INFO] grid plugin loaded.
[2019-07-10 02:35:07,284] [INFO] Session-stats plugin loaded.
[2019-07-10 02:35:07,284] [INFO] [update] plugin loaded.
[2019-07-10 02:35:07,285] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
[2019-07-10 02:35:07,285] [INFO] initializing waveshare v1 2.9 inch display
[2019-07-10 02:35:07,286] [DEBUG] [led] event 'loaded' set
[2019-07-10 02:35:07,287] [DEBUG] [led] using pattern 'oo  oo  oo oo  oo  oo  oo' ...
```

After disabling the display in the config pwnagotchi will start just fine.

```
ui:
    display:
      enabled: false
      type: 'waveshare29inch'
      color: 'black'
```

**Environment (please complete the following information):**
 - Pwnagotchi version 1.4.1
 - OS version MacOS
 - Type of hardware PI4 + PI Zero
 - Any additional hardware used

Any idea?

moheshmohan, 2019-12-15:
Hmm this looks like similar issue i faced with 2.13 inch display. The display becomes frozen and ends up freezing pwnagotchi too. So i found its an issue with the original python driver from waveshare itself

So i ended up writing a freeze recovery routine in the original waveshare code. You can see it here

#720 

Refer the last commits on this pull request to see the anti freeze code. Bascially in the busy loop it will try to reinitialize itself by turning off and on again. You could try same approach to solve the issue. There is very little documenation that i could understand from waveshare wiki so all this is based on reading their python code and repurposing it. Hope this helps

ack202, 2019-12-15:
Thanks for the hint. Used your code but display is still black. Will investigate further.

moheshmohan, 2019-12-15:
That code is for 2.13 inch b&c variants of display, you'll have to check offical waveshare epaper repo and improvise it for 2.9inch display...i think 2.9inch also has a, b, c and d variants so u will have to work one based on ur variant

ack202, 2019-12-17:
Got it running with the new drivers for the 29bc model.
After each refresh the display flickers on/off for about 10 times. Did you run into something similar?

moheshmohan, 2019-12-18:
I didnot see flicker for 10 times but may be 3 or 4 on 2.13bc. But yes there is that flicker on refresh

darkr4y, 2020-01-16:
I hv same issue with 213bc and cant figure it out, plz help, this wut have I done.
```
0. install with latest github release
1. sudo systemctl stop pwnagotchi
2. cd /usr/local/src/pwnagotchi && git pull
3. sudo pip3 install -r requirements.txt
4. sudo python3 setup.py install
5. sudo cp -r pwnagotchi/ /usr/local/src/pwnagotchi/build/scripts-3.7/pwnagotchi
6. sudo vim /etc/pwnagotchi/config.yml  #change the type to "waveshare_213bc"
```
lso got stuck after running sudo pwnagotchi --debug
```
copying /usr/local/lib/python3.7/dist-packages/pwnagotchi/defaults.yml to /etc/pwnagotchi/default.toml ...
[2020-01-16 16:54:57,558] [DEBUG] loading plugins from /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default - enabled: ['grid', 'auto-update', 'led', 'session-stats']
[2020-01-16 16:54:57,567] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/grid.py
[2020-01-16 16:54:57,575] [DEBUG] loaded plugin grid as <grid.Grid object at 0xb4e033b0>
[2020-01-16 16:54:57,581] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/session-stats.py
[2020-01-16 16:54:57,587] [DEBUG] loaded plugin session-stats as <session-stats.SessionStats object at 0xb4e03250>
[2020-01-16 16:54:57,592] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/auto-update.py
[2020-01-16 16:55:00,339] [DEBUG] loaded plugin auto-update as <auto-update.AutoUpdate object at 0xb4e037b0>
[2020-01-16 16:55:00,344] [DEBUG] loading /usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/default/led.py
[2020-01-16 16:55:00,353] [DEBUG] loaded plugin led as <led.Led object at 0xb57d3fb0>
[2020-01-16 16:55:00,374] [INFO] Session-stats plugin loaded.
[2020-01-16 16:55:00,377] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
[2020-01-16 16:55:00,381] [INFO] [update] plugin loaded.
[2020-01-16 16:55:00,379] [INFO] grid plugin loaded.
[2020-01-16 16:55:00,388] [DEBUG] [led] event 'loaded' set
[2020-01-16 16:55:00,392] [WARNING] ui.fps is 0, the display will only update for major changes
[2020-01-16 16:55:00,402] [INFO] initializing waveshare v2 display
[2020-01-16 16:55:00,412] [DEBUG] [led] using pattern 'oo  oo  oo oo  oo  oo  oo' ...
```

moheshmohan, 2020-03-10:
Try changing the display type to waveshare213bc on new version pwnagotchi