evilsocket/pwnagotchi #729: Add bt-nap to pwnagotchi iso

H4ck3r404, 2019-12-14:
I think that this could some of that bluetooth problems:

http://edwardsh.in/pihud/2013/01/25/tethering-the-raspberry-pi-with-an-android-phone-over-bluetooth

(Also include more specific instructions for connecting.)

dadav, 2019-12-16:
We have another repository for the documentation. If you have some suggestions for better instructions, please create an issue there. I don't know what you mean with "Add bt-nap to pwnagotchi iso". The required bluetooth packages are already installed. The connection is established via the plugin. If you install another tool, the plugin should be disabled. But the official way is to use the plugin.

H4ck3r404, 2019-12-18:
The problem is that official plugin can't connect to some od the devices. Like my phone for example: I was connecting it manualy with all possibile ways, even so after succesfull pairing it immidieately disconnects. Command:

connect <mac>

shows bluetooth error. And I'm not alone with this problem. (See #377.)

crysalis010, 2020-01-04:
Exactly the same problem here as  @H4ck3r404 

With bt-nap installed, I can connect to the pwnagotchi, see the web interface, etc, but no internet connection sharing with iOS 13 device.

Ended up disabling the feature until it is worked out.

H4ck3r404, 2020-01-04:
@crysalis010 was your phone's bluetooth tethering enabled while pairing? If it wasn't then you must reflash image. Pwnagotchi will never create bt-tether after pairing without bluetooth tethering enabled. It was said somewere on project page. I think that this should be marked in the future.

H4ck3r404, 2020-01-04:
Creating config file properly also could be the problem. My only working setup was with pairing as I mentioned before with:

share_internet: true

dadav, 2020-04-11:
Derek close

closed 2020-04-11 by derek[bot]