evilsocket/pwnagotchi #731: [BUG] Update breaks the brain

hmax42, 2019-12-15:
**Describe the bug**
updating from 1.3.0 to 1.4.1 leads to:
`[2019-12-15 20:10:51,983] [INFO] [ai] loading /root/brain.nn ...
[2019-12-15 20:10:52,259] [ERROR] error while starting AI
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/__init__.py", line 51, in load
    a2c.load(config['path'], env)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/base_class.py", line 700, in load
    model.set_env(env)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/base_class.py", line 96, in set_env
    "Error: the environment passed must have at least the same observation space as the model was trained on."
AssertionError: Error: the environment passed must have at least the same observation space as the model was trained on.`

**To Reproduce**
this behaviour was observed via:
* backing up a 1.3.0 brain and restoring it to a fresh 1.4.1 image
* updating a fresh 1.3.0 image restored with a 1.3.0 backup via git/pip to master (?)

**Environment**
* Raspberry Pi 3 A+

**Expected behavior**
no error message and loading of the AI after updating

hmax42, 2019-12-15:
now i can even find the error even in a fresh 1.3.0 with restored backup.

Evg33, 2019-12-16:
`rm /root/brain.nn
`

hmax42, 2019-12-16:
tried that. didn't get recreated.
hmm, maybe i did not wait long enough...
will try

Evg33, 2019-12-16:
@hmax42 wait 10 minutes ....

hmax42, 2019-12-17:
so
1. attached it in manual mode
2. stop the service
3. rename brain.nn to brain.nn.old
4. force start auto mode
5. after 2 minutes my raspi a3+ started training and failed

```
[2019-12-17 17:00:44,876] [INFO] entering auto mode ...
...
[2019-12-17 17:02:45,787] [INFO] [ai] model created:
[2019-12-17 17:02:45,788] [INFO]       gamma: 0.99
[2019-12-17 17:02:45,789] [INFO]       n_steps: 1
[2019-12-17 17:02:45,790] [INFO]       vf_coef: 0.25
[2019-12-17 17:02:45,791] [INFO]       ent_coef: 0.01
[2019-12-17 17:02:45,792] [INFO]       max_grad_norm: 0.5
[2019-12-17 17:02:45,793] [INFO]       learning_rate: 0.001
[2019-12-17 17:02:45,794] [INFO]       alpha: 0.99
[2019-12-17 17:02:45,795] [INFO]       epsilon: 1e-05
[2019-12-17 17:02:45,796] [INFO]       verbose: 1
[2019-12-17 17:02:45,797] [INFO]       lr_schedule: constant
[2019-12-17 17:02:45,798] [INFO]       tensorboard_log: None
[2019-12-17 17:02:45,837] [INFO] [ai] learning for 50 epochs ...
[2019-12-17 17:02:45,852] [ERROR] [ai] error while training
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 231, in learn
    runner = A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 310, in __init__
    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/runners.py", line 19, in __init__
    self.obs[:] = env.reset()
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
    self._save_obs(env_idx, obs)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
    self.buf_obs[key][env_idx] = obs
```

dadav, 2019-12-17:
Could you pls try it with the latest code?

Do this to update:

```shell
cd /usr/local/src/pwnagotchi
git pull
pip3 install .
```

hmax42, 2019-12-17:
i already did.
that was the way i updated

`...updated to version 1.4.1`
the "1.4.1" part is now obviously wrong. i updated it as you wrote it.

hmax42, 2019-12-19:
git pull just set the new version number today, but i got an extra line in the log
maybe i overlooked it before

```
[2019-12-19 20:56:41,763] [ERROR] [ai] error while training
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/ai/train.py", line 177, in _ai_worker
    self._model.learn(total_timesteps=epochs_per_episode, callback=self.on_ai_training_step)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 231, in learn
    runner = A2CRunner(self.env, self, n_steps=self.n_steps, gamma=self.gamma)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/a2c/a2c.py", line 310, in __init__
    super(A2CRunner, self).__init__(env=env, model=model, n_steps=n_steps)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/runners.py", line 19, in __init__
    self.obs[:] = env.reset()
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 52, in reset
    self._save_obs(env_idx, obs)
  File "/usr/local/lib/python3.7/dist-packages/stable_baselines/common/vec_env/dummy_vec_env.py", line 71, in _save_obs
    self.buf_obs[key][env_idx] = obs
ValueError: could not broadcast input array from shape (428) into shape (1,503)
```

xBelladonna, 2019-12-22:
I have been able to reproduce this issue on my unit, in fact I could not figure out why it was happening or trigger it to happen reliably. I became frustrated and erased my SD card, reinstalling fresh 1.4.1 image, boot in manual mode to stop all pwnagotchi-related services by systemd, then update each component by hand separately. Then upon second boot in auto mode, baby starts working properly! 

I know this is not either a graceful or entirely logical solution but I thought it useful to share since I had same problem and aforementioned procedure fixed it for me.

hmax42, 2019-12-27:
i reset the pi 3a+ now with just a fresh 1.4.3 image and the old config.yml.
still the same error (same numbers)

i start to believe the pi 3 a+ is to blame.
but i am pretty sure it worked with the early images.

more testing

hmax42, 2019-12-28:
success with a clean 1.0.1 image and this config.yml

```
main:
    plugins:
      grid:
        enabled: false
      auto-update:
        enabled: false
personality:
    deauth: false
ui:
    display:
        enabled: true
        rotation: 180
        type: 'waveshare_1'
```

Commander-Crash, 2020-01-01:
-- Logs begin at Wed 2020-01-01 08:06:25 GMT. --
Jan 01 08:18:42 Keram0n pwnagotchi-launcher[749]: [2020-01-01 08:18:42,647] [WARNING] From /usr/local/lib/python3.7/dist-packages/tensorflow_estimator/python/estimator/api/_v1/estimator/__init__.py:10: The name tf.estimator.inputs is deprecated. Please use tf.compat.v1.estimator.inputs instead.
Jan 01 08:19:37 Keram0n pwnagotchi-launcher[749]: [2020-01-01 08:19:37,878] [INFO] [epoch 4] duration=00:01:15 slept_for=00:01:00 blind=0 inactive=5 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=96% mem=50% temperature=45C reward=-0.2
Jan 01 08:20:52 Keram0n pwnagotchi-launcher[749]: [2020-01-01 08:20:52,152] [INFO] [epoch 5] duration=00:01:14 slept_for=00:01:00 blind=0 inactive=6 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=96% mem=50% temperature=45C reward=-0.2
Jan 01 08:22:08 Keram0n pwnagotchi-launcher[749]: [2020-01-01 08:22:08,245] [INFO] [epoch 6] duration=00:01:16 slept_for=00:01:00 blind=0 inactive=7 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=97% mem=50% temperature=45C reward=-0.2
Jan 01 08:23:07 Keram0n pwnagotchi-launcher[749]: [2020-01-01 08:23:07,251] [WARNING] !!! captured new handshake on channel 11, -82 dBm: 98:b8:ba:29:cf:85 () -> WIN_502536 [fc:2b:b2:3e:9a:02 ()] !!!
Jan 01 08:23:10 Keram0n systemd[1]: Stopping pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning....
Jan 01 08:23:10 Keram0n systemd[1]: pwnagotchi.service: Main process exited, code=killed, status=15/TERM
Jan 01 08:23:10 Keram0n systemd[1]: pwnagotchi.service: Succeeded.
Jan 01 08:23:10 Keram0n systemd[1]: Stopped pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..
Jan 01 08:23:10 Keram0n systemd[1]: Started pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning..
Jan 01 08:23:23 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:23,654] [INFO] memtemp plugin loaded.
Jan 01 08:23:23 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:23,666] [INFO] Session-stats plugin loaded.
Jan 01 08:23:23 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:23,669] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
Jan 01 08:23:23 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:23,671] [INFO] grid plugin loaded.
Jan 01 08:23:23 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:23,730] [WARNING] ui.fps is 0, the display will only update for major changes
Jan 01 08:23:23 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:23,746] [INFO] initializing waveshare v2 display
Jan 01 08:23:26 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:26,116] [INFO] [ai] loading /root/brain.json
Jan 01 08:23:26 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:26,195] [INFO] Keram0n@ecaae949f513d80c80bc31dd9493bdccee10a43418c16d647594b418e86be3e6 (v1.4.1)
Jan 01 08:23:26 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:26,211] [INFO] entering auto mode ...
Jan 01 08:23:26 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:26,244] [INFO] [ai] bootstrapping dependencies ...
Jan 01 08:23:26 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:26,421] [INFO] web ui available at http://0.0.0.0:8080/
Jan 01 08:23:27 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:27,521] [INFO] connecting to http://localhost:8081/api ...
Jan 01 08:23:29 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:29,927] [INFO] found monitor interface: mon0
Jan 01 08:23:29 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:29,946] [INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14]
Jan 01 08:23:29 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:29,980] [INFO] handshakes will be collected inside /boot/handshakes
Jan 01 08:23:48 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:23:48,933] [INFO] [epoch 0] duration=00:00:22 slept_for=00:00:00 blind=0 inactive=1 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=97% mem=30% temperature=45C reward=-0.2
Jan 01 08:24:39 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:24:39,305] [WARNING] From /usr/lib/python3/dist-packages/tensorflow/__init__.py:98: The name tf.AUTO_REUSE is deprecated. Please use tf.compat.v1.AUTO_REUSE instead.
Jan 01 08:24:39 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:24:39,320] [WARNING] From /usr/lib/python3/dist-packages/tensorflow/__init__.py:98: The name tf.AttrValue is deprecated. Please use tf.compat.v1.AttrValue instead.
Jan 01 08:24:39 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:24:39,326] [WARNING] From /usr/lib/python3/dist-packages/tensorflow/__init__.py:98: The name tf.COMPILER_VERSION is deprecated. Please use tf.version.COMPILER_VERSION instead.
Jan 01 08:24:39 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:24:39,332] [WARNING] From /usr/lib/python3/dist-packages/tensorflow/__init__.py:98: The name tf.CXX11_ABI_FLAG is deprecated. Please use tf.sysconfig.CXX11_ABI_FLAG instead.
Jan 01 08:24:39 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:24:39,337] [WARNING] From /usr/lib/python3/dist-packages/tensorflow/__init__.py:98: The name tf.ConditionalAccumulator is deprecated. Please use tf.compat.v1.ConditionalAccumulator instead.
Jan 01 08:25:15 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:25:15,097] [INFO] [ai] creating model ...
[... cut ...]
Jan 01 08:31:27 Keram0n pwnagotchi-launcher[2860]:     "Error: the environment passed must have at least the same action space as the model was trained on."
Jan 01 08:31:27 Keram0n pwnagotchi-launcher[2860]: AssertionError: Error: the environment passed must have at least the same action space as the model was trained on.
Jan 01 08:31:27 Keram0n pwnagotchi-launcher[2860]: [2020-01-01 08:31:27,570] [WARNING] [ai] AI not loaded!

This issue just started on a Raspberry pi Zero W. Using v1.4.1 on Buster. My logs if this helps anyone.

Commander-Crash, 2020-01-01:
This issue was fixed for me after removing brain.nn from /root

Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,423] [INFO] [ai] model created:
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,438] [INFO]       gamma: 0.99
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,442] [INFO]       n_steps: 1
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,452] [INFO]       vf_coef: 0.25
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,456] [INFO]       ent_coef: 0.01
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,467] [INFO]       max_grad_norm: 0.5
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,471] [INFO]       learning_rate: 0.001
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,482] [INFO]       alpha: 0.99
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,485] [INFO]       epsilon: 1e-05
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,496] [INFO]       verbose: 1
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,501] [INFO]       lr_schedule: constant
Jan 01 09:58:25 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:25,536] [INFO]       tensorboard_log: None
Jan 01 09:58:26 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:26,099] [INFO] [ai] learning for 50 epochs ...
Jan 01 09:58:54 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:58:54,620] [INFO] [epoch 18] duration=00:01:05 slept_for=00:01:00 blind=0 inactive=13 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=63% mem=60% temperature=52C reward=-0.1368421052631579
Jan 01 09:59:08 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:08,841] [INFO] [ai] setting new policy:
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,009] [INFO] [ai] ! min_rssi: -200 -> -70
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,176] [INFO] [ai] ! ap_ttl: 120 -> 341
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,215] [INFO] [ai] ! sta_ttl: 300 -> 141
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,265] [INFO] [ai] ! recon_time: 30 -> 52
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,339] [INFO] [ai] ! max_inactive_scale: 2 -> 6
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,380] [INFO] [ai] ! recon_inactive_multiplier: 2 -> 3
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,404] [INFO] [ai] ! hop_recon_time: 10 -> 12
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,435] [INFO] [ai] ! min_recon_time: 5 -> 30
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,446] [INFO] [ai] ! max_misses_for_recon: 5 -> 7
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,489] [INFO] [ai] ! excited_num_epochs: 10 -> 14
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,520] [INFO] [ai] ! bored_num_epochs: 15 -> 7
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,536] [INFO] [ai] ! sad_num_epochs: 25 -> 6
Jan 01 09:59:09 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:09,570] [INFO] [ai] ! channels: [] -> [1, 3, 4, 9, 10, 13, 14]
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,075] [INFO] [ai] --- training epoch 1/50 ---
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,082] [INFO] [ai] REWARD: -0.136842
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,105] [INFO] [ai] observation:
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,135] [INFO]     aps
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,139] [INFO]       CH 1: 0.3163265306119221
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,148] [INFO]       CH 2: 0.010204081632642649
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,160] [INFO]       CH 4: 0.010204081632642649
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,177] [INFO]       CH 5: 0.010204081632642649
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,193] [INFO]       CH 6: 0.15306122448963974
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,208] [INFO]       CH 8: 0.020408163265285297
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,229] [INFO]       CH 9: 0.030612244897927948
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,233] [INFO]       CH 10: 0.010204081632642649
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,243] [INFO]       CH 11: 0.43877551020363387
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,247] [INFO]     sta
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,258] [INFO]       CH 6: 0.4374999999972657
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,262] [INFO]       CH 8: 0.06249999999960938
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,275] [INFO]       CH 11: 0.49999999999687506
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,286] [INFO]     peers
Jan 01 09:59:10 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:10,308] [INFO] [ai] saving model to /root/brain.nn ...
Jan 01 09:59:17 Keram0n pwnagotchi-launcher[6448]: [2020-01-01 09:59:17,338] [INFO] [ai] saving /root/brain.json

mgagnequebec, 2020-03-03:
I also have problems with the AI on r3b+, I installed the latest release and did the update "pip3 install .".
Manu mode works, Auto mode also. But the AI never kicks in (always stuck in AUTO mode). It's been running for 6 hours now, still in AUTO mode :(

dadav, 2020-04-11:
Derek close

closed 2020-04-11 by derek[bot]

qkum, 2021-03-18:
I can confirm 18-3-2021 that the fix is to remove the brain.nn file -> reboot -> it recreates it when the AI is done loading. You can easily track it on the log plugin in the browser. 

Just encountered this problem within 24 hours after installing pwnagotchi on my Raspberry pi 4 with the official Pwnagotchi 1/2/3 image. 

**Command to delete the file:**
_sudo rm /root/brain.nn_

Then reboot