evilsocket/pwnagotchi #737: Some typos in voice and onlinehashcrack & paw-gps plugin

xBelladonna, 2019-12-17:
- There are some typos in the voice (and translations) and the onlinehashcrack plugin. This is very minor nitpick but probably important to have proper spelling.
- PAW GPS plugin is needing IP and port in format `ip:port`, unclear in default configuration
- There is also no announcement in logs of the onlinehashcrack plugin on loaded unlike all other plugins which add entry to inform they are loaded.
- Finally, this is not a problem, but onlinehashcrack plugin does not instruct pwnagotchi to change face expression when uploading, face inherit whatever else is changing it and only status text is placed.

I propose to fix the typos, add log entry by onlinehashcrack plugin on loaded and add cute face when onlinehashcrack plugin is uploading handshakes.

dadav, 2019-12-17:
Hi, did you follow this guide?

https://pwnagotchi.ai/contributing/#updating-an-existing-translation

Because the `mo` files are missing. So your changes won't have any effect

xBelladonna, 2019-12-17:
I'm an idiot, I did not commit them, thank you for telling me. I have pushed new commit to PR addressing this.

closed 2020-01-19 by xBelladonna