evilsocket/pwnagotchi #743: [BUG]Auto-Update plugin not updating pwnagotchi

benallard, 2019-12-20:
**Describe the bug**
The auto-update plugins keeps rebooting my gotchi (once an hour), but doesn't update it.

**To Reproduce**
1. Install older image
2. give internet to your gotchi
3. see it reboot
4. again and again ...
5. again and again ...

**Expected behavior**
Reboot once, enjoy the new stuff.

LoganMD, 2019-12-21:
Pretty sure everyone is having this issue, me included

dadav, 2019-12-21:
Which older image?

benallard, 2019-12-21:
1.3.0 and 1.4.1 had the issue.

th0rtz, 2019-12-21:
Same problem in 1.3.0

Error log on update : 
>[2019-12-21 13:37:09,525] [WARNING] update for evilsocket/pwnagotchi available (local version is '1.3.0'): https://github.com/evilsocket/pwnagotchi/archive/v1.4.2.zip
>[2019-12-21 13:37:11,251] [INFO] [update] downloading https://github.com/evilsocket/pwnagotchi/archive/v1.4.2.zip to /tmp/updates/pwnagotchi/pwnagotchi_1.4.2.zip ...
sh: 0: getcwd() failed: No such file or directory
> Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory

> [2019-12-21 13:37:32,028] [INFO] [update] extracting /tmp/updates/pwnagotchi/pwnagotchi_1.4.2.zip to /tmp/updates/pwnagotchi ...

dadav, 2019-12-21:
1.3.0 is known to have an update-bug. But I thought 1.4.1 is fine :/

xBelladonna, 2019-12-22:
Hmm, funny you should mention that, I have experienced update problems on 1.4.1 too, mentioned in my comment on #731, but I'm not sure if it's the same issue as described here. Anyway, I stopped services by systemd and updated all components manually as a workaround.

LoganMD, 2019-12-22:
Having update problems too, just won't update from 1.4.1 to 1.4.2. Had the same problem with 1.3.0.

th0rtz, 2019-12-23:
Manual update work fine from 1.3.0 to 1.4.2

a60wattfish, 2019-12-23:
The downloader for the update isn't following the redirect to the download file, so just saves the HTML header instead of the zip.

phr34k0, 2019-12-28:
Looks like it needs to do some version checking on that line 214 instead of just

> if num_installed > 0:

```
def on_internet_available(self, agent):
.
.
                if num_installed > 0:
                    display.update(force=True, new_data={'status': 'Rebooting ...'})
                    pwnagotchi.reboot()

            except Exception as e:
                logging.error("[update] %s" % e)
```

dadav, 2020-01-16:
So, I think it's fixed, if not we will reopen.

dadav, 2020-01-16:
Derek close

closed 2020-01-16 by derek[bot]

alessaba, 2020-03-18:
> #731

Can you pleas explain what you mean by "updating all the components manually"?
do you clone the project to the pi and then overwrite the Pwnagotchi folders?
thanks in advance

phr34k0, 2020-03-18:
> > #731
> 
> Can you pleas explain what you mean by "updating all the components manually"?
> do you clone the project to the pi and then overwrite the Pwnagotchi folders?
> thanks in advance

Nope. Delete everything and throw your pi out the window

alessaba, 2020-03-18:
> > > #731
> > 
> > 
> > Can you pleas explain what you mean by "updating all the components manually"?
> > do you clone the project to the pi and then overwrite the Pwnagotchi folders?
> > thanks in advance
> 
> Nope. Delete everything and throw your pi out the window

I just wanted to know what folders i need to overwrite. Thanks anyway for the professional behavior 👍

phr34k0, 2020-03-18:
> I just wanted to know what folders i need to overwrite. Thanks anyway for the professional behavior

LOL. The answer is written all over above. Yes. Manual is delete everything and git clone.