evilsocket/pwnagotchi #748: Integrate dnsmasq to give DHCP lease on usb0

xBelladonna, 2019-12-22:
Issue #481 was raised to use a DHCP server to configure network for usb0 when in manual mode instead of having to manually configure the adapter.

I have would like to submit a pull request to address this issue, but before I open it, I must discuss some points.

This PR would:
  - Installs and configures dnsmasq to offer DHCP on usb0
  - Sets DHCP options 3 (routers) and 6 (nameservers) to nothing so we don't try to route the host's traffic through Pwnagotchi!
  - Sets the bootp-dynamic option so that annoying Windows clients will accept the DHCPOFFER because apparently unless you give non-empty DHCP option 3 it won't send back DHCPACK......
  - Sets the DHCP lease time to 1m (I believe it ends up at 2 minutes due to DHCP standard anyway) as a workaround for MAC address of ethernet gadget changing with every reboot.
    - This makes sure when Pwnagotchi is rebooted, the old DHCP lease is cleared ASAP and host can get DHCP lease again. 
    - Downside is that reboot takes faster than this, so host most likely will timeout when doing DHCPDISCOVER, and it is the case in my testing, however if connection is removed and replugged (with external power so it doesn't die) after this time, DHCP lease has expired and host will get IP again. Very ungraceful solution, however user can also renew DHCP lease manually if they know how, also not a real solution IMO.

Only idea I had so far is symlinking `/var/lib/misc/dnsmasq.leases` to `/dev/null`. This makes leases never remembered but even that doesn't work, I think dnsmasq is starting too late after ethernet gadget becomes ready and DHCP client on Windows at least is already timing out and default to APIPA addressing by then...

If anyone has suggestions on how to solve this problem on Pwnagotchi (rather not adjust DHCP timeout on host, almost impossible on Windows without advanced knowledge) please comment! I would really like this to Just Work:tm:

xBelladonna, 2019-12-22:
First mistake, MUST configure DHCP lease time of 2 minutes otherwise lease time defaults to infinity!
Second, it seems all this problem about lease times/symlinking leases file to oblivion can be solved by [this community hack in the documentation](https://pwnagotchi.ai/community/#static-rdnis-gadget-to-avoid-reconfiguration-everytime-you-plug-it-to-the-computer)

dadav, 2020-01-16:
Derek close

closed 2020-01-16 by derek[bot]