evilsocket/pwnagotchi #756: Bluetooth connection without requiring tether

RChadwick7, 2019-12-26:
While there appear to be many phones that do not allow Bluetooth Tethering (None of my devices do), and Pwnagotchi doesn't need Internet for someone to log into the interface or terminal, it would be great if Bluetooth could be made to work without tethering.

gtsatsis, 2019-12-26:
@RChadwick7, As far as I'm aware, Bluetooth Tethering is the only way to enable IP communication through Bluetooth.

Sure, internet access isn't required but the internet protocol is.

RChadwick7, 2019-12-26:
What about a Bluetooth PAN?

gtsatsis, 2019-12-26:
@RChadwick7, Yeah, but that's what is carrier-locked on many phones because Plans are usually able to share the host connection.

Afaik, it's just what Bluetooth Tethering does, makes a PAN & Shares host.

foreign-sub, 2019-12-28:
Hi @RChadwick7 ,
As @gtsatsis said, the issue here is that without bluetooth tethering your phone does not add the PAN to the bluetooth profile sent to your pwnagotchi, i don't know of any workaround, sorry.

PS: Always enable bluetooth tethering before pairing otherwise the bluetooth profile will miss the PAN  even if you enable it after.

RChadwick7, 2019-12-28:
I wish I could enable Bluetooth tethering, but my phone doesn't have that option. Apparently T-Mobile in the US likes to disable it. My Galaxy S7, Galaxy S10, and an LG Tablet all are missing the Bluetooth Tethering option.
Do you know if Pwnagotchi would work with Pdanet?

dadav, 2020-01-16:
If you can't tether, you won't be able to create a connection :(

dadav, 2020-01-16:
Derek close

closed 2020-01-16 by derek[bot]