evilsocket/pwnagotchi #760: Make the URL/Endpoint of the net-pos module configurable

DaniloNC, 2019-12-27:
**Is your feature request related to a problem? Please describe.**

Currently, the net-pos module has the Mozilla geolocation API URL hardcoded on the plugin configuration.

```
MOZILLA_API_URL = 'https://location.services.mozilla.com/v1/geolocate?key={api}'
 ```
It would be good if this endpoint could be edited in the configuration file, thus allowing to use the google geolocate API: https://developers.google.com/maps/documentation/geolocation/intro

Both APIs are compatible and google geolocate API seems to be more accurate than Mozilla's in my tests.

The downside of Google API is that it is a paid API (although, you have a 200USD credit every month that should be enough to a bunch of pwnagotchis for free. 200USD=40000 geolocate requests)

**Describe the solution you'd like**

Make the MOZILLA_API_URL parameter configurable (and potentially change the variable name)

**Describe alternatives you've considered**

Creating a custom plugin using the google maps API should be easy, but I think more people would benefit from having it configurable.

jacopotediosi, 2020-02-14:
Hi, thanks for your good work!
How to manually update this plugin before relase?
I updated file inside /usr/local/src/pwnagotchi/pwnagotchi/plugins, but I see in logs that it still loads the old original plugin

dadav, 2020-04-05:
Derek close

closed 2020-04-05 by derek[bot]