evilsocket/pwnagotchi #764: Change the way pwnagotchi can get online.

Commander-Crash, 2019-12-29:
I just installed and set up on a rpi zero all is going well for the most part. But what really bugs me is how it is set up for internet access. I mean WTF you ppl thinking? I spent a full day trying to get this on line. Im a advance user in linux and windows. I have a piratebox that can go online with a few lines in the resolv,conf file. Why can't this be the same? I know i may be a bit harsh and maybe I'm missing something but jesus christ I felt like I installed windows ME and booted for the 1st time. Please change this or point me in the right direction to change it myself. I'm using your pre built .IMG running Buster.

phr34k0, 2019-12-29:
Pwnagotchi has a community and we all contribute. Personally I didn't like how messy most of the scripts was coded with undefined exception errors, simple gui positioning to avoid overlap and downloading cracked APs from WPA-SEC in a single text file without extracting each found lines to it's respective .gps.cracked so it could be parsed by the WEBGPSMAP plugin.

Instead of complaining do your own change and contribute if you're that said 'advance user' :P

xenDE, 2019-12-29:
> downloading cracked APs from WPA-SEC in a single text file without extracting each found lines to it's respective .gps.cracked so it could be parsed by the WEBGPSMAP plugin.

feel free to add this to the plugin, i run it only on boot:
https://github.com/xenDE/pwnagotchi-tools/blob/master/wpa-sec.founds.insert.py

phr34k0, 2019-12-29:
> feel free to add this to the plugin, i run it only on boot:
> https://github.com/xenDE/pwnagotchi-tools/blob/master/wpa-sec.founds.insert.py

Yeah I did a merge to your script not too long ago. Just made a new issue to yours. Thanks xen!

dadav, 2019-12-29:
Obviously a troll post. I'll close this.

dadav, 2019-12-29:
Derek close

closed 2019-12-29 by derek[bot]