evilsocket/pwnagotchi #768: pwnagotchi's web UI refuses the connection

FerdinandoPH, 2019-12-30:
When I try to connect to the web UI, I get an error saying that the pwnagotchi refused the connection. Why does this happen?

H4ck3r404, 2019-12-31:
What kind od setup are you using?

FerdinandoPH, 2020-01-05:
I'm using a Raspberry Pi 4B, with a 16GB SD card.

H4ck3r404, 2020-01-05:
Does error ocured during pairing by bluetoothctl?
What kind od error was it?
And can you post your config file?

H4ck3r404, 2020-01-05:
Post your config file and details about connection. (ssh and page that you were connecting to)

FerdinandoPH, 2020-01-06:
Iḿ trying to connect to http://pwnagotchi.local:8080/
This is my config.yml
`main:
	name: 'pwnagotchifph'
	whitelist:
		- 'WIFI_FERINAL_CENTRAL'
		- 'WIFI_FERINAL_CENTRAL_PLUS'
		- 'WIFI_FERINAL'
		- 'vodafone E3D4'
	plugins:
		gps:
			enabled: true
		auto-update:
			enabled: true
		memtemp:
			enabled: true
		net-pos:
			enabled: true
		onlinehashcrack:
			enabled: true
		quickdic: 
			enabled: true
			wordlist_folder: /home/pi/Probable-Wordlists-master
		webcfg:
			enabled: true
		bt-tether:
			enabled: true
			devices:
				Oneplus6TFPH:
					enabled: true
					search order: 1
					mac: '64:a2:f9:f4:fe:4a'
					ip: '192.168.44.44'
					netmask: 24
					interval: 1
					scantime: 15
					share_internet: true
					priority: 99
					max_tries: 5
		webgpsmap:
			enabled: true
		grid:
			enabled: true
			report: true
			exclude:
				- 'vodafoneE3D4'

ui:
	web:
		enabled: true
		username: pi
		password: raspberry`

H4ck3r404, 2020-01-07:
Try connecting to 192.168.44.44:8080

phr34k0, 2020-01-13:
I had the same issue while I was abroad and it works when I was back in my own country. I get that "connection error" that prints on the page instead of the webui. Apparently solved it with having Zerotier installed, turning on bridging on my home computers and changing my DNS to that of Google DNS (8.8.8.8, 8.8.4.4) worked. 

Got to know that the place I was at has DNS resolving issues that caused the internet itself to be super slow while browsing some other sites.

dadav, 2020-04-11:
Derek close

closed 2020-04-11 by derek[bot]