evilsocket/pwnagotchi #773: Torrent's for Downloading.

Commander-Crash, 2020-01-01:
Have your image/zips on a torrent tracker. Github amazon servers are atrocious and cuts off with out the ability to resume. anything over 100mb's should not be hosted by github.

Comodore125, 2020-01-01:
github files on amazon s3 are slow for me too. I had speeds less than 5% of my downlink.

AliceGrey, 2020-01-02:
@Commander-Crash @Comodore125 @evilsocket I'd be willing to upload new versions to linuxtracker.org if someone wants to seed the torrent from a VPS/seedbox/etc

roens, 2020-02-21:
Could also be uploaded to the [Archive](https://archive.org), where uploaded files are available via torrent…

virtadpt, 2020-02-21:
Keep in mind that, even though the Internet Archive offers torrents, whether or not they seed those torrents is hit or miss (I've been fighting this particular battle for a while, the ratio is something like 1/11 in favor of "no").

Workflow for this is upload to the Archive.  When it's up grab the torrent and throw it on a seedbox.  Start torrent.  If the download's not done inside of an hour, stop (not delete) the torrent.  Download all of the files from the Archive and put them into the torrent's download directory.  Then start the torrent back up and you'll be the (one) seeder.

FractumSeraph, 2020-03-03:
I've packaged the current release into a torrent. I'll be hosting it on my seedbox for exactly one month.

[.Torrent File](https://github.com/evilsocket/pwnagotchi/files/4283645/Pwnagotchi.1.4.3.torrent.zip)

[Magnet Link](magnet:?xt=urn:btih:b62605d7dbd641274a9c84d56fb10cbfa7aa9cb2&dn=Pwnagotchi+1.4.3&tr=udp://glotorrents.pw:6969/announce&tr=udp://tracker.opentrackr.org:1337/announce&tr=udp://torrent.gresille.org:80/announce&tr=udp://tracker.openbittorrent.com:80&tr=udp://tracker.coppersurfer.tk:6969&tr=udp://tracker.leechers-paradise.org:6969&tr=udp://p4p.arenabg.ch:1337&tr=udp://tracker.internetwarriors.net:1337&tr=wss://tracker.openwebtorrent.com)