evilsocket/pwnagotchi #778: Can pwnagotchi attack JUST specific networks?

braimee, 2020-01-06:
I'm wondering if there's a config option/tweak I can make such that Pwnagotchi *only* grabs handshakes from SSIDs I specify.  

## Expected Behaviour

Much like the *whitelist* option creates a list of networks to ignore, it would be nice to have a parameter like *only_attack* which would by default only grab handshakes from SSIDs specified in the config file.  This way, if I'm heading out to a customer and going to do a wireless assessment as part of the deal, they can tell me their SSIDs are Corp1, CorpGuest and CorpInternal, and I can load those into the *only_attack* setting in the Pwnagotchi so I don't get into hot water be attacking other SSIDs in the space.

## Current Behaviour

Pwnagotchi pwns all the things.

## Possible Solution

I suppose I could audit the network space ahead of time, create a list of IDs and then try to whitelist as many as possible to limit Pwnagotchi's pwnage, but probably not super practical :-)

## Steps to Reproduce (for bugs)

1.
2.
3.
4.

## Context

## Your Environment

- [ ] You're using the official images

- [ ] You're using a raspberry pi 0

- [ ] You're using a supported LCD

dadav, 2020-01-07:
I think this is not so easy to implement and a too big contrast to the normal usecase.

dadav, 2020-01-09:
Derek close

closed 2020-01-09 by derek[bot]