evilsocket/pwnagotchi #780: Much needed plugins and features. My experience with Pwnagotchi.

CyrisXD, 2020-01-07:
Firstly let me say, this is an amazing little device and is super fun to play with. However considering it's nature it's a very "noisy" device and approach.

&nbsp;

As we go about our lives we're often taking the same routes to get somewhere, or going into corporate settings, 
cities etc... By doing this we are capturing an enormous amount of handshakes, but here's the problem.  

&nbsp;

The Pwnagotchi can and will constantly deauth clients even if a handshake was already captured. It will also keep associating to the same networks over and over again and each time it captures the handshake or PMKID for that SSID it overwrites the file. 

&nbsp;

It's essentially creating a very noticeable attack being client side. Think of devices that are streaming content like a Chromecast, as soon as it looses connection the streaming stops and shows an error. If this repeatedly happens for someone they are going to know something is up.

&nbsp;

**There's a few features I wish the Pwnagotchi had, that maybe some plugin developers can work on.**

&nbsp;

* Pwnagotchi should automatically whitelist SSID's once a complete handshake has been captured. So to ignore those networks and not keep deauth'ing those clients.

* The ability to quickly disable/enable deauth with a [button like the plugins panel](https://i.imgur.com/caKBEcH.png). (Less hassle than entering the webcfg and trying to search all those settings on a small screen like your mobile). This way we can easily disable deauth if we're going into somewhere we'd like to be a little quieter. 

* Some kind of handshake stats panel to show which handshakes were captured on which dates. To sort them and download only the applicable ones, not the old handshakes. This can help you keep track of what's new, what you're done with etc...You can already download handshakes directly with the handshakes-dl plugin, however the ability to delete handshake files would also be greatly useful. Especially if the SSID has automatically been added to the whitelist it won't add the file back into the directory. This will keep your handshake directory much cleaner.  You should also be able to remove the SSID from the whitelist here too.

* It should be more clear what type of info was captured in the PCAP, often running the conversion tools I don’t easily know which one to use. If the PCAP has the PKMID or a handshake. Append capture file name with capture type, PMKID or HS. ex. “MyWiFi-as2X2-PMKID.pcap”.

&nbsp;

I'd be interested in hearing your thoughts. Projects like this make me wish I learned Python instead of taking the Javascript route, otherwise I'd be building all these. 

&nbsp;

A big thank you to @evilsocket for this project and I can't wait to see what it develops into. 

&nbsp;

Cheers

phr34k0, 2020-01-13:
Agreed though I pretty much like that 

> The Pwnagotchi can and will constantly deauth clients even if a handshake was already captured. It will also keep associating to the same networks over and over again and each time it captures the handshake or PMKID for that SSID it overwrites the file.

"feature".. It pretty much helps to do a correction if it constantly deauth and resaves a new GPS.JSON file after a handshake with current GPS issue on either stuck values with "Latitude": 0, "Longitude": 0 if it has problems getting a GPS fix through the GPS/GNSS dongle or hat, or if it fails to capture any GPS if you either walk, drive too fast or forget to turn on PAW-GPS.

To sum it up, GPS part is still wonky. Until the day GPS feature gets stable and fixed with a plugin that one could use a 4G/GSM/GPRS/GNSS hat addon, or a feature to grab off positioning data from your phone bluetooth by celltowers instead of a slow GPS Fix that most people in the community are using with the WebGPSMap plugin

virtadpt, 2020-01-13:
Is there a design reason that gpsd doesn't appear to be an option?

Junohea, 2020-01-14:
For these 2 requests
> - Pwnagotchi should automatically whitelist SSID's once a complete handshake has been captured.
> - It should be more clear what type of info was captured in the PCAP, often running the conversion tools I don’t easily know which one to use. If the PCAP has the PKMID or a handshake. Append capture file name with capture type, PMKID or HS. ex. “MyWiFi-as2X2-PMKID.pcap”.

I started a plugin to help with hash/handshake related validation, check it out [here ](https://github.com/evilsocket/pwnagotchi-plugins-contrib/blob/master/hashie.py)
Every time a new packet is added to a pcap it tries to extract a PMKID & EAPOL hash from it, then write that to a file. 
Those written files (which indicate a "complete handshake") could be leveraged as a check. I'll look into what i could add that might help, but it would likely be a toggle that the user would enable to say "yes please add these to a the whitelist". I just dont know the limitations yet, like would the config need to be reloaded to take effect and how cumbersome the appending to the whitelist is.

The only downside is, once you capture enough and start whitelisting a network, if they change the network's password you won't continue to get deauth packets without removing the network from the whitelist (although you would still capture PMKIDs).

qkum, 2021-03-18:
> For these 2 requests
>
>
> I started a plugin to help with hash/handshake related validation, check it out [here ](https://github.com/evilsocket/pwnagotchi-plugins-contrib/blob/master/hashie.py)
> Every [...]

People never change the password.

Only companies with a professional IT department. 

I'm yet to ever irl see a wifi password get changed.

Schools, gymnasiums, universities, small companies...never.

It never changes after the day it is first set.

phr34k0, 2021-03-19:
> People never change the password.
> 
> Only companies with a professional IT department change passwords maybe once a year.
> 
> I'm yet to ever irl see a wifi password get changed.
> 
> Schools, gymnasiums, universities, small companies...never.
> 
> It never changes after the day it is first set.

Probably that only applies to you. I personally have had Residential Access Points and CCTVs that changed their password like some of the guys down here. Just leave it at that for when the Router and MAC addresses are the same and the possibility of owners changing their passwords either by choice to change it, automated by software or someone else having access, messed with it or changed it thus forcing the owner to change their password via a reset, ISP Technician or firmware flash.

I have had Residential Access point having a change MAC address and Password but with the same SSID at some point only because I own 2.4 ghz - 5 ghz military jammers with me that I jammed the whole apartment I live in from time to time. This made the access point owner to change their entire rig since inevitably even if they reset or have physical access.