evilsocket/pwnagotchi #782: [BUG & SOLUTION] Raspberry Pi & Pwnagotchi 11 Channels by Default

phr34k0, 2020-01-13:
**Describe the bug**
I have noticed that by default all the images downloadable from this git release has Raspberry Pi to only Channel 11. Pwnagotchi will utilize all that Channels that are shown in the iwlist if you had your "channels:" under personality in config.yml file left blank.. ex. "channels: [] "

**To Reproduce**
Steps to reproduce the behavior:
1. `iwlist wlan0 channel` or `iwlist wlan0 channel` in terminal to check all supported channels

**Solution**
Edit your `nano /etc/rc.local` file and add `sleep 3 && iw reg set SG` above exit 0. Replace SG with your own country code.

Example:

`#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

# BEGIN ANSIBLE MANAGED BLOCK
if ! /opt/vc/bin/tvservice -s | egrep 'HDMI|DVI'; then
  /opt/vc/bin/tvservice -o
fi
# END ANSIBLE MANAGED BLOCK
sleep 3 && iw reg set NO
exit 0`

hectorm, 2020-01-13:
If the `crda` package is installed I think it would be better to modify the `/etc/default/crda` file.

phr34k0, 2020-01-13:
> If the `crda` package is installed I think it would be better to modify the `/etc/default/crda` file.

Well if that works and is loaded right after boot before pwnagotchi gets loaded that would be a good solution instead of rc.local. Have not tested that one out yet.

closed 2020-01-16 by phr34k0