evilsocket/pwnagotchi #793: Timeout Communicating with api.pwnagotchi.ai

kruffin, 2020-01-15:
The pwnagotchi is plugged in via USB and has internet shared, but in the pwngrid-peer.log file it shows a timeout when attempting to pull the inbox and enrolling.

## Expected Behaviour
I would expect it to not timeout trying to communicate with the api.pwnagotchi.ai site.

## Current Behaviour
It times out with this line in the logs:

    POST https://api.pwnagotchi.ai/api/v1/unit/enroll (1m0.001496378s) Post https://api.pwnagotchi.ai/api/v1/unit/enroll: net/http: request canceled (Client.Timeout exceeded while awaiting headers)
    GET https://api.pwnagotchi.ai/api/v1/unit/inbox/?p=1 (1m0.000902714s) Get https://api.pwnagotchi.ai/api/v1/unit/inbox/?p=1: net/http: request canceled (Client.Timeout exceeded while awaiting headers)

A curl to the inbox without auth directly on the pi gets an unauthorized response, so it would seem connectivity is there.

    $ curl https://api.pwnagotchi.ai/api/v1/unit/inbox/?p=1
    {"error":"unauthorized"}

## Possible Solution

## Steps to Reproduce (for bugs)
Not sure how to reproduce, but have done a fresh install of 1.4.3 and restored from backup with the same issue before and after the backup/restore.

## Context

## Your Environment

- [X] You're using the official images (1.4.3)

- [X] You're using a raspberry pi 0

- [X] You're using a supported LCD (Waveshare v2)

Have a realtime clock (pcf8523) module attached and setup per the adafruit instructions.

kruffin, 2020-01-15:
Additional logs from pwnagotchi.log:

> [2020-01-15 18:22:36,626] [ERROR] [grid] error while checking inbox: (status 422) {"error":"Get https://api.pwnagotchi.ai/api/v1/unit/inbox/?p=1: net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}
> [2020-01-15 18:23:36,833] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:24:37,904] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:25:39,034] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:26:40,185] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:27:41,314] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:28:42,506] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:29:43,684] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
> [2020-01-15 18:29:56,375] [ERROR] [grid] error while checking inbox: (status 422) {"error":"Post https://api.pwnagotchi.ai/api/v1/unit/enroll: read tcp 10.0.0.2:49092-\u003e104.24.119.236:443: read: connection timed out"}

natur3kid, 2020-01-15:
Same error here, when updating from 1.4.3

kruffin, 2020-01-15:
Hmm, I've found this thread: https://community.pwnagotchi.ai/t/pwnagotchi-stopped-reporting-pwned-networks-pwngrid-error/466
I'll try removing the duplicates in the .api-report.json file like suggested and report back here.

kruffin, 2020-01-15:
After doing that and rebooting, I am seeing more error information:

> [2020-01-15 21:13:47,445] [INFO] grid: parsing /root/handshakes/WFMINF_f07f06c0ab7a.pcap ...
> [2020-01-15 21:13:47,538] [ERROR] grid: Could not find field [BSSID]
> [2020-01-15 21:14:47,606] [ERROR] error while reporting ap WFMINF(f0:7f:06:c0:ab:7a)
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/grid.py", line 100, in report_ap
>     'bssid': bssid,
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/grid.py", line 36, in call
>     raise Exception("(status %d) %s" % (r.status_code, r.text))
> Exception: (status 422) {"error":"Post https://api.pwnagotchi.ai/api/v1/unit/report/ap: net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}

So maybe this is it not being able to pull the BSSID from the pcap? Some pcaps didn't have this error, but a lot are.

kruffin, 2020-01-15:
Running the extract_from_pcap() manually indicates it couldn't pull the BSSID from the pcap. Wonder if that logic isn't correct.

> from pwnagotchi.utils import WifiInfo, extract_from_pcap
> out = extract_from_pcap('/root/handshakes/WFMINF_f07f06c0ab7a.pcap', [WifiInfo.BSSID, WifiInfo.ESSID])
> reading from file /root/handshakes/WFMINF_f07f06c0ab7a.pcap, link-type IEEE802_11_RADIO (802.11 plus radiotap header)
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/utils.py", line 296, in extract_from_pcap
>     raise FieldNotFoundError("Could not find field [BSSID]")
> pwnagotchi.utils.FieldNotFoundError: Could not find field [BSSID]

kruffin, 2020-01-15:
Rebooted and still get the same inbox errors. It did upload new networks after fixing the api-report.json file so that's one win. Not sure what the repercussions of this (the inbox 422 error) particular error are though. 

There are definitely some threading issues since after a while of uploading pcaps it started getting errors trying to create a thread:

> [2020-01-15 21:53:03,241] [ERROR] can't start new thread
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 78, in one
>     _thread.start_new_thread(locked_cb, locked_cb_args)
> RuntimeError: can't start new thread
> [2020-01-15 21:53:03,244] [ERROR] error while running led.on_internet_available : can't start new thread
> [2020-01-15 21:53:03,297] [ERROR] can't start new thread
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 78, in one
>     _thread.start_new_thread(locked_cb, locked_cb_args)
> RuntimeError: can't start new thread
> [2020-01-15 21:53:08,516] [ERROR] error while running grid.on_internet_available : can't start new thread
> [2020-01-15 21:53:08,518] [ERROR] can't start new thread
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 78, in one
>     _thread.start_new_thread(locked_cb, locked_cb_args)
> RuntimeError: can't start new thread
> [2020-01-15 21:53:08,522] [ERROR] error while running auto-update.on_internet_available : can't start new thread
> [2020-01-15 21:53:08,524] [ERROR] can't start new thread
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 78, in one
>     _thread.start_new_thread(locked_cb, locked_cb_args)

closed 2020-04-13 by evilsocket

RoiArthurB, 2020-04-24:
Since I updated my pwnagotchi to version 1.5.3 I have the same exact error :
Log : 
```
[2020-04-24 04:07:22,801] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
[2020-04-24 04:08:25,258] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
[2020-04-24 04:09:27,563] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
[2020-04-24 04:10:29,801] [ERROR] [grid] error while checking inbox: HTTPConnectionPool(host='127.0.0.1', port=8666): Read timed out. (read timeout=60.0)
[2020-04-24 04:12:33,140] [ERROR] [grid] error while checking inbox: (status 422) {"error":"Get https://api.pwnagotchi.ai/api/v1/unit/inbox/?p=1: net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}
[2020-04-24 04:13:33,529] [ERROR] [grid] error while checking inbox: (status 422) {"error":"Get https://api.pwnagotchi.ai/api/v1/unit/inbox/?p=1: net/http: request canceled (Client.Timeout exceeded while awaiting headers)"}
```

And curl :
```
$ curl https://api.pwnagotchi.ai/api/v1/unit/inbox/\?p\=1
{"error":"unauthorized"}
```

balicali420, 2023-08-04:
im on v1.5.5 and still having this

virtadpt, 2023-09-06:
The API has been dead for a couple of months, at least.

PR #845 (2020-04-13): ready to merge

- fixed imports (especially if you just want to get the version or work with plugins)
- fixed chrome issue (fixed #819)
- fixed the output of some error msgs
- fixed config output (sections are now separated by newline)
- fixed thread problem (fixed #793)
- added sad/bored states to reward function (fixed #51)
- added on_config_changed cb (fixed #822)
- added reward curve to session-stats plugin
- added logtail plugin
- added event polling via websocket (fixed #43)
- added plugins subcommand (fixed #105)
- added remove_whitelisted function (closes #734)
- added whitelist functi [... cut]