evilsocket/pwnagotchi #801: Let plugins change the Pwnagotchi face

emedvedev, 2020-01-18:
**Is your feature request related to a problem? Please describe.**

I would like the plugins to be able to change the Pwnagotchi face, as they add quite a bit of functionality, but the face stays the same. Sometimes it creates a weird mismatch.

**Describe the solution you'd like**

One way to implement this was proposed in #440: `pwnagotchi.ui.faces` is loaded into the plugin file, and then faces are updated from the config (in case there were custom modifications) in the following way:

```
config = agent.config()
faces.load_from_config(config['ui']['faces'])
```

Since most plugin events expose the `agent` object, there are no core modifications necessary, and the only files that need to be modified are the plugin files. I think it's the least intrusive solution.

If not all plugin events expose `agent`, I can also create a PR to expose it, if necessary. 

Let me know if that's an agreeable solution, or if there's anything else you'd prefer! I'll be happy to work on it.

dadav, 2020-01-20:
But then you also have to change them back to the normal ones! When do you do that? Plugins are running all the time, even in parallel, so I think this might be a problem...

emedvedev, 2020-01-20:
Pwnagotchi events don't return the face to a "normal" state, it just gets replaced with a next face when the next event triggers, so the plugins doesn't have to either. They get replaced just like before—as a response to the various [view.py events](https://github.com/evilsocket/pwnagotchi/blob/0764304be9b69b4db8f925f6a97750193148c5b9/pwnagotchi/ui/view.py#L138-L150).

There's also (`on_normal`)[https://github.com/evilsocket/pwnagotchi/blob/0764304be9b69b4db8f925f6a97750193148c5b9/pwnagotchi/ui/view.py#L179] that [triggers on every association](https://github.com/evilsocket/pwnagotchi/blob/0764304be9b69b4db8f925f6a97750193148c5b9/pwnagotchi/agent.py#L393), so a face never holds for too long.

emedvedev, 2020-01-20:
Basically, it's not much different from changing and updating the status text, and the plugins do that all the time without issues. The only difference is that the faces have to be loaded from the config first.

dadav, 2020-01-20:
You are right, they normaly don't reset the faces, but if you load the plugin-faces with `faces.load_from_config`, the normal faces get orverwritten, therefore the events in `view.py` will also use the plugin faces.

emedvedev, 2020-01-20:
Wait, but isn’t that the point of loading the faces from config in the first place? The normal ones should be overwritten if custom faces are specified in the config file. `view.py` does the same during init.

dadav, 2020-01-20:
Maybe I don't get your point xD Lets say every plugin has a own set of faces. Each plugin gets called if certain events are happening, e.g. a handshake is captured. Plugin_1 changes the faces, does something. Plugin_2 changes the faces, does something, Plugin_3.....and so on. When every plugin got called, another handshake gets captured and this process repeats. The real faces (set in config.toml) never actually will be used anymore, right?

Btw. whats wrong with just do:
```python
ui.set('face', '^.^')
```
when you want to change the face?

emedvedev, 2020-01-20:
Ah, no, that’s not what I meant at all, sorry for the confusion! I meant that I want to be able to use the faces specified in config.toml in plugins. `ui.set('face', '^.^')` works, but `ui.set('face', faces.HAPPY)` doesn’t, because `faces` isn’t exposed to plugins.

This is exactly what my proposal (and the PR I referred to) is about: import faces into plugins and load custom faces (if any) from the config. Then they can be used in plugins with `ui.set`. You can take a look at the PR in the issue description :) It was rejected because it was considered a core change, I’m just not completely sure what would be evilsocket’s preferred way to implement this.

dadav, 2020-01-20:
Hm, aren't they exposed to the plugins? You can just do this:

```python
from pwnagotchi.ui import faces

def on_the_hook_you_want(self, ui):
  ui.set('face', faces.HAPPY)
```

You don't need to call `load_from_config` in the module, because they are called when pwnagtochi starts...

emedvedev, 2020-01-20:
I’ll need to check that, actually. But that’s pretty much what my PR did, and it got rejected.

dadav, 2020-01-20:
His main critic was that you didn't discussed it first... The community.pwnagotchi.ai site would be the right place for this^^ I think it's a good idea, so if you want to do another PR (without the load_from_config stuff), I can test if it's working as expected and see if we can get it to be merged.

emedvedev, 2020-01-20:
That's exactly why I've opened the issue :) I'll take a stab at another PR and test if it works without load_from_config (you're right, it looks like it should). Thanks!